RetroPie/RetroPie-Setup #1481: neogeo documentaion

davidt99, 2016-05-13:
After following the mame tutorial, and playing with clrmamepro and neogeo bios, I found out a couple of things:

- The [neogeo list](https://github.com/RetroPie/RetroPie-Setup/wiki/MAME#bios) is incomplete. Few roms are missing (like "uni-bios_3_2.rom" and sfix.**sfx**), some same roms has different file extension (sp-j2.rom and sp-j2.sp1).

- 000-lo.lo has two versions: one with crc of 5a86cff2 and one with e09e253c.

I found out all of this by examining the neogeo specification in the .dat file of each emulator and looking at the clrmamepro's log file.
If it's ok by you, I can create a pull request and change the documentation to include all the files and warn the user about the two versions of 000-lo.

dankcushions, 2016-05-13:
The neogeo bios roms changes for different emulators. for example, I know that current FBA (lr-fba-next) recently had some bios changes to neogeo, so I wonder if we should even list the contents at all, unless ALL versions of MAME used by retropie use the exact same neogeo files, which would need investigation.

When rebuilding romsets it should get build a specific neogeo.zip anyway.

davidt99, 2016-05-13:
I already invastigated all the emulators and tried to make "Super" neogeo bios, but the 000-lo.lo issue prevented this.
I agree that when rebuilding a romesets, it should get build with specific neogeo bios, but the simple user doesn't know it - he will download the zip from one of the websites and will fail to rebuild the roms (or succeed if he got lucky with his version), even if he looked at the current documentation.

I just want to make the documentation clearer, that all.

joolswills, 2016-05-16:
The documentation is on a Wiki and can be improved by anyone with a github account. If unsure, you can use the retropie forum to discuss changes https://retropie.org.uk/forum/

closed 2016-05-16 by joolswills