RetroPie/RetroPie-Setup #1484: Request for mp3 support in lr-picodrive / current source install fails

BenWlson, 2016-05-15:
Request that lr-picodrive be compiled with mp3 support enabled. With the use of mp3 audio compression, a complete SegaCD - US romset can actually be installed in a reasonable amount of  space.

I tried adding libavcodec-dev and installing from source, but picodrive is unusable afterwards. See [upstream  issue](https://github.com/libretro/picodrive/issues/24)
Here is my build log: http://pastebin.com/Yqm5YJ5h
It looks like the compile may using Rpi-1 settings. I'm not 100% sure if this is purely an upstream issue or if it's something that can be corrected in the retropie install script.

Either way, I'd like the mp3 request/issue to remain as a placeholder until the compile issues can be resolved.

BenWlson, 2016-05-17:
I have had no luck getting mp3 support to work in Picodrive, but I may have figured out why it's not compiling properly.

Our script uses:
`make -f Makefile.libretro platform=raspberrypi`

Looking at that portion of Makefile.libretro, I find:
`CFLAGS += -marm -mfpu=vfp -mfloat-abi=hard -march=armv6j`

I was able to get picodrive to compile from source and work by either changing our script to remove the platform parameter or by editing the makefile to remove the -march parameter. Not sure which would be the preferred method.

In my attempts to get MP3 support, I tried installing libavcodec-dev in hopes that it would autodetect it during the configure - didn't work. I tried editing the configure file to change the default result of have_avcodec to yes instead of autodetect - didn't work.

Picodrive should support MP3 files, I just can't figure it out.

InsecureSpike, 2016-05-17:
are you sure it's supported in this particular port of picodrive we are actually using?
as from what I remember some features are missing

BenWlson, 2016-05-17:
I don't know for certain. There appears to be some code for it.

InsecureSpike, 2016-05-17:
I had a kinda similar issue regarding support for the Sega Pico, the emulator was not supported in this port.....
could be the same issue

InsecureSpike, 2016-05-17:
http://libretro.com/forums/showthread.php?t=734

joolswills, 2016-05-20:
I can't find any reference to the mp3 support in the libretro makefile - so I'm pretty sure it isn't supported in the libretro core. This should be reported/requested upstream.

closed 2016-05-20 by joolswills

joolswills, 2016-05-20:
I have submitted a fix for the undefined symbol upstream, although I will also commit a change so we build differently on rpi1 vs rpi2/3 (using C z80 library on rpi2/rpi3 for compatibility - drz80 is old arm code)