RetroPie/RetroPie-Setup #1487: Shutdown Broken After Kernel 4.4 Update.

opensourcefan, 2016-05-22:
After updating Kernel 4.4 alone (my case) or installing the latest update or an install via the current image, pi hangs on boot up after the shutdown is used. Regardless if it's shut down via EmulationStation or via command line or a halt command is sent via SSH the pi will not boot and goes to a black screen. For me unplugging during the hang and plugging back in will allow it so start normally. If I do not use the shutdown and just unplug it will boot just fine. 

Using 3.7, I only installed kernel 4.4 and did not install the full update.

joolswills, 2016-05-22:
Doesn't happen here - RPI shutdowns and reboots as expected. There would be far more users reporting it if there was a major crash on shutdown issue - but you can try and report it to the raspberry firmware issues page - https://github.com/raspberrypi/firmware - RetroPie is not an OS, and we are not responsible for the firmware - RetroPie image is just Raspbian with RetroPie installed on top.

Also in your thread above, you say it used to shutdown with the green light on - this is something that was fixed in a firmware update on the rpi3 (ie, now it correctly goes off)

joolswills, 2016-05-22:
BTW, it is normal to need to remove power after shutting down to start up again. if you want to restart you should try the "reboot" command (or the reboot option from Emulation Station)

closed 2016-05-22 by joolswills

opensourcefan, 2016-05-22:
Not trying to reboot, just happens after turning it back on after shutting down. I understand about it not being an OS, I'm in the wrong spot.

joolswills, 2016-05-22:
Did you try another sdcard ? might be related to your sdcard and firmware / driver changes.

dankcushions, 2016-05-22:
Define 'hangs'? Mine takes around a minute to save all the metadata (normal), then the signal to the TV stops, but the pi's green/red lights will remain on, but it's off, effectively. To reboot, you have to cycle the power. This is all normal (apart from possibly the green light remaining on, which apparently I could fix by updating the firmware)

opensourcefan, 2016-05-22:
When the shutdown command is used, the screen goes black after about 5 seconds. If can then see the green led flash and do its thing. It eventually goes out and only the red stays on. Then I unplug. 

Upon plugging back in when ever that is the screen stays black and the green light pulses. Nothing happens no matter how long I leave it. Pulling the plug and putting it back in makes it boot normally.

joolswills, 2016-05-22:
What PSU ? Did you try another sdcard yet ?

opensourcefan, 2016-05-22:
I did try another card with an image that wasn't updated (early 3.7 build), it worked as it should.

I'm using a 3000ma power adapter.

joolswills, 2016-05-22:
Do you have USB devices plugged in ? If so, have you tested without ?

joolswills, 2016-05-22:
I mean to try another sdcard with the latest updates.

opensourcefan, 2016-05-22:
I'll create a fresh install on another card and see what happens, with the current updates, great idea. 

I have no USB devices plugged in, just a wireless PS3 controller.

joolswills, 2016-05-22:
Try it without the PS3 controller paired also.

opensourcefan, 2016-05-22:
I did let the controller timeout and turn off last night. I sent a halt command from my phone but it did the same thing.

joolswills, 2016-05-22:
It's worth trying another PSU/USB cable also in case it is power related. I can't really offer any more advice. I suspect it's something specific to your set-up or there would be far more people with the problem.

opensourcefan, 2016-05-22:
I understand, thank you so much for all this advice. Other than that it's an excellent setup that I have, I may just live with it if I can't fix it.

opensourcefan, 2016-05-22:
Fixed:

Okay so in my case I had just upgraded the kernel to 4.4, this was before the whole dist-upgrade was out. Today I did the full dist-upgrade, downgraded my bluetooth to 1.0 and shutdown appears to be working again.

In order this is what I did:

`sudo apt-get update && sudo apt-get dist-upgrade`
`sudo reboot`
`wget http://archive.raspberrypi.org/debian/pool/main/p/pi-bluetooth/pi-bluetooth_0.1.0_armhf.deb`
`sudo dpkg -i pi-bluetooth_0.1.0_armhf.deb`
`sudo reboot`

After that the shutdown worked just fine.