RetroPie/RetroPie-Setup #1493: AMIGA controls differences between RPi1 and RPi2/3

padremayi, 2016-05-27:
The hardcoded controls is different from RPi1 and RPi3 (or 2).

With RPi3 (or 2) the button for emulator menu is F12
With RPi1 the button for emulator menu is CTRL SX

With RPi1 the hardcoded buttons don't work, they work only with RPi3 (or 2)

joolswills, 2016-05-27:
uae4arm has been updated recently and you probably have an older version running on the rpi1 machine. Did you update both ?

padremayi, 2016-05-27:
For sure on RPi3 I have the latest version, installed yesterday evening. But as I said before I have to use F12 in order to enter UAE4arm menu.

On RPi1 I'll check the version

joolswills, 2016-05-27:
That is correct. It switched to F12.

closed 2016-05-27 by joolswills

padremayi, 2016-05-27:
Update the AMIGA page on GitHub

joolswills, 2016-05-27:
Feel free. It's a wiki.

padremayi, 2016-05-27:
Ok, is there another command to be updated?

joolswills, 2016-05-27:
not sure what you mean ?

padremayi, 2016-05-27:
Another control that is reported incorrect on the wiki

joolswills, 2016-05-27:
The wiki has documentation on uae4all - not sure it really covers uae4arm currently. So if you are going to edit it, you need to make a new section for uae4arm

HerbFargus, 2016-05-27:
From home page of the wiki 
```
This wiki is supposed to be created by the users of the RetroPie setup script for the users of the script. Everyone can contribute to it!
```

padremayi, 2016-05-27:
Ok but the controls are hardcoded, who codes them?

joolswills, 2016-05-27:
The emulator author/porter for uae4arm -

https://github.com/Chips-fr/uae4arm-rpi

joolswills, 2016-05-27:
This would be best continued on the forum - we prefer to use this issue tracker for bugs only (which is why it asks to use the forum before opening a ticket)

padremayi, 2016-05-27:
Ok thanks, I was not able to find this type of informations on UAE4arm, now it's clear ;-)