RetroPie/RetroPie-Setup #1495: Is there a way to remap keyboard? Xmodmap?

padremayi, 2016-05-28:
Hi to all, I have a problem with my keyboard because RetroPie doesn't recognize function keys.

I need F12 in order to enter UAE4arm menù and exit emulator, is there a way to remap F12, or other function key, to a combination of others keys? A system like Xmodmap...

Thanks

joolswills, 2016-05-28:
Not sure what you mean by RetroPie doesn't recognise function keys - retropie is a system to install emulators and consists of many components. You should use the forum for support as mentioned when opening an issue. 

>Please make sure you read the "guidelines for contributing" above before opening an issue. Support questions should go to the forum. Please post to the forum first, as many "issues" can be solved with configuration changes. Please remove this text before posting.

closed 2016-05-28 by joolswills

padremayi, 2016-05-28:
I mean that when I press a function key like F12 RetroPie doesn't do nothing

joolswills, 2016-05-29:
RetroPie itself doesn't use the f12 key. But please continue this on the forum - https://retropie.org.uk/forum/ where you can get help on the emulator you are using. thanks.

padremayi, 2016-05-29:
Yes, I already wrote there. Could you say me if RetroPie/EmulationStation use X11?

joolswills, 2016-05-29:
Use the forum.