RetroPie/RetroPie-Setup #1499: Slight problem in rewinder

matalog, 2016-05-31:
I have noticed that the rewinder in retroarch on retropie doesn't act the same as the rewinder on retroarch on a PC or fceux's lua rewinder on pc.

I noticed this holding a shell and rewinding in SMB 3 on NES but i checked and it's the same in SMW on SNES. If you rewind holding a shell on retropie, you will let go of the shell wether or not you are holding B down when you are finished rewinding. If you hold B while rewinding on a PC, in retroarch (at least using fceumm) you will hold onto the shell when you are finished rewinding.

This suggests that the rewinder releases B upon finishing rewinding, maybe only for a small length of time, but long enough to register and make you drop the shell.

This can be a problem when experimenting in these type of games - obviously the original creators of the rewind feature took care that is wasn't like that on PC.

Kind regards,

Matt.

joolswills, 2016-06-01:
Did you post to the forum first as requested when opening an issue ?

The only way to resolve this is to have the hotkey as an unused button in the game. Normally RetroArch blocks input when the hotkey is pressed, which requires it to be a different key other than select (it works in some cases, but it is impossible to insert credits on some emulators if select is used). We have a workaround in RetroPie to fix this, but it does mean when select is initially pressed, it will be sent to the game first - which I assume in this game will cause that to happen.

joolswills, 2016-06-01:
I'm assuming it is related to this anyway, but in any case, I'm not sure there is anything we can do about it - if you can reproduce it on a non RetroPie retroarch on the Raspberry Pi (ie without our change mentioned above) you can report it to the RetroArch project.

matalog, 2016-06-01:
I posted it to the link below and it was ignored.

https://retropie.org.uk/forum/topic/1584/rewinder-in-retroarch-isn-t-normal

I'll give what you said a go, i think you could be correct, it is about select.

Date: Wed, 1 Jun 2016 14:39:13 -0700

joolswills, 2016-06-01:
Sometimes stuff get missed due to the quantity of posts (or at least I don't have time to read everything). You could bump a post if people miss it - also use tags and it will be easier for people to find when searching etc.

matalog, 2016-06-01:
I have just tried changing the hotkey activation button to R2 and my rewind button is L2 - this has the same problem.

I have had more than one issue ignored - i assumed it was due to the technicality of my questions!

Date: Wed, 1 Jun 2016 14:50:35 -0700

joolswills, 2016-06-01:
did you read https://retropie.org.uk/forum/topic/3/read-this-first - the more information you provide, the more likely you are to get help - you didn't provide much of the requested info.

matalog, 2016-06-02:
At the start of talking with me about this, you seemed to know what i was talking about.  

Current version 3.8 - any controller.

joolswills, 2016-06-03:
I am not referring to this issue, but your comment regarding not getting replies on the forum. I checked some of your forum posts, and you do not provide that much information to go on, which is why you maybe didn't get an answer hence me posting you the forum sticky which gives more information on what should be provided.

joolswills, 2016-06-03:
I have tested this without our custom RetroArch changes and I can reproduce it - but you will need to report it to the RetroArch project as it's not a fault with RetroPie itself. https://github.com/libretro/RetroArch - pressing the hotkey enable button causes the B button to stop being pressed I assume.

joolswills, 2016-06-04:
Upstream have confirmed this is by design. Closing.

closed 2016-06-04 by joolswills