RetroPie/RetroPie-Setup #1502: Configure Bluetooth Devices breaks RPI3 onboard Bluetooth

aleehedl, 2016-06-01:
Opening option Configure Bluetooth Devices in Retropie-Setup installs package **bluetooth** which apparently somehow conflicts with existing bluetooth packages (pi-bluetooth?). The same is true when installing the PS3 driver.

By removing the **bluetooth** dependency in `scriptmodules/supplementary/bluetooth.sh` and in `scriptmodules/supplementary/ps3controller.sh` and uninstalling package **bluetooth** I was able to get my bluetooth and PS3 controller working again (even if the pairing process still is a bit tricky and requires me to manually run sixad --start).

joolswills, 2016-06-01:
the ps3 package when installed disables the bluetooth, but if you install the bluetooth after it may not - they don't play nicely together - it is a known issue - once we have a reliable way of pairing ps3 controllers without sixad we will switch away from it.

joolswills, 2016-06-01:
https://github.com/RetroPie/RetroPie-Setup/issues/1181

aleehedl, 2016-06-01:
Ok. The whole procedure could be documented slightly better. Should the sixad be started automatically by Retropie somehow?

My setup now works with pi-bluetooth package installed **without** installing the bluetooth. I start the sixad in rc.local.

joolswills, 2016-06-01:
it is mention at least on https://github.com/RetroPie/RetroPie-Setup/wiki/Setting-up-a-Bluetooth-controller - the documentation is user contributed - feel free to improve it.

closed 2016-06-01 by joolswills