RetroPie/RetroPie-Setup #1504: ES Theme: Simple crashes my system...

RetroPieNerd, 2016-06-03:
I know that Simple has a "white screen" problem but, I have Pi 3 and I bumped up the GPU RAM to 256 megabytes but it still did nothing. Then, I bumped it up to 500 megabytes, this time it was a little better..
It did not completely crash my system but I could still do nothing but cold boot it and unplug it. I have no idea if it has to do with my latest contributions to the theme or what...?

HerbFargus, 2016-06-03:
If your svgs were invalid then it will crash but it will hardcore and show an error and freeze the framebuffer. Also if your images weren't optimised then the vram has a nasty habit of causing your white screen of death. You're starting to realise why we created carbon to replace the simple theme ;) I only ever got 15 or less systems for the simple theme and even then it sometimes pushed it. The pi just doesn't have the processing for it with the way it was coded.

Remove a bunch of systems and see if that improves anything

RetroPieNerd, 2016-06-03:
Aww.. Too bad, that was my favorite theme along with the Pixel theme! Guess I will have to stick with the Pixel theme instead...

Thanks anyways! :)

closed 2016-06-03 by RetroPieNerd