RetroPie/RetroPie-Setup #1505: x86 setup: "sh: 1: retroarch: not found"

AxxlForce, 2016-06-03:
after executing retropie_setup.sh on a x86 Ubuntu 16.04 machine and doing an install from source that's what i get when trying to load a core from emulationstation.

retroarch is installed in /opt/retropie/emulators/retroarch . if i make it available via PATH in .bashrc I can run "retroarch" from command line but it doesn't detect any cores.

joolswills, 2016-06-03:
I will need a copy of the build log from RetroPie-Setup/logs

joolswills, 2016-06-03:
and a copy of `~/emulationstation/es_log.txt` please and `/tmp/runcommand.log` (after launching something)

AxxlForce, 2016-06-03:
i've searched my entire hdd there is no runcommand.log. but then again I'm unable to launch something in the first place.

Here's the logs: 
https://docs.google.com/uc?id=0BwToJlEUVDK1V0NHLXZ3ZnBVdXM&export=download

joolswills, 2016-06-03:
You have made a manual/custom emulationstation configuration via es_systems.cfg ? RetroPie does not use this

```
retroarch --fullscreen -L /usr/lib/libretro/fceumm_libretro.so PATH_TO_ROM
```

but launches via runcommand

joolswills, 2016-06-03:
You have either edited `/etc/emulationstation/es_systems.cfg` or you have an older/own config in `~/.emulationstation/es_systems.cfg`

AxxlForce, 2016-06-04:
that's true i did create my own ~/.emulationstation/es_systems.cfg as i thought that was mandatory. allrite, will removing ~/.emulationstation do the trick then? unfortunatelly i can't check back until monday. thanks for the help so far!

HerbFargus, 2016-06-04:
See the guide here: when installing from the setup script the es_systems.cfg will be generated automatically. as this issue tracker is for bugs not configuration issues, please continue any discussion on the forum as requested initially when you first opened this issue.

closed 2016-06-04 by HerbFargus