RetroPie/RetroPie-Setup #1511: Better mupen64plus defaults

loganmc10, 2016-06-07:
As you are probably aware, many users have issues getting N64 emulation to works properly, here are my suggestions:

Make mupen64plus-GLideN64 the default, not gles2rice. Honestly if it was me, I would get rid of gles2rice, gles2n64, and the videocore plugin. GLideN64 works betters than all those in every game I have tried. Just leave mupen64plus-GLideN64 and lr-mupen64plus.

Make CEA-1 the default resolution for N64 games. If you don't use CEA-1 then mupen64plus is going to try to upscale the image to 1080p (assuming they are using an HDTV for instance). That creates big slowdowns for people.

Set gpu_freq=500 in config.txt (at least for the rpi3). This doesn't void the warranty and seems quite stable, it is also needed for a lot of N64 games to work well.

I would also increase the gpu_mem amount. It looks like it's setting ```gpu_mem_1024``` to 256. I would probably increase that to around 500MB-600MB. GLideN64 incorporates a texture cache, and the default texture cache max size is 500MB, so if someone plays for a long time there is the possibility for that cache to fill up.

joolswills, 2016-06-07:
@gizmo98 thoughts ?

Regarding resolution - we could try and force that res, but you assume all screens will support it . runcommand does have some code for this to try and pick a working res, but it's not ideal - eg if you are on a small screen etc it may not work at all.

we are definitely not overclocking by default. This should be up to end users.

Increasing GPU mem may upset other users too, I think our current defaults are ok.

joolswills, 2016-06-07:
I don't want to remove choice/configurability either - this is what RetroPie is all about. You may not think those plugins are needed, but I can assure you there would be great upset by many if they were removed.

loganmc10, 2016-06-07:
@joolswills all good points.

If you aren't going to increase the gpu_mem then you might want to consider changing the default texture cache size in GLideN64.

in mupen64plus.cfg under ```[Video-GLideN64]``` the default is:
```
# Size of texture cache in megabytes. Good value is VRAM*3/4
CacheSize = 500
```

loganmc10, 2016-06-07:
Also, in regards to the resolution. There is another way!

I submitted a patch to the SDL developers here:
https://bugzilla.libsdl.org/show_bug.cgi?id=3353

Which would allow the Raspberry Pi to stretch a window to fit the full screen, with that patch you can run mupen64plus with: ``--windowed --resolution 640x480``` and it would run mupen64plus at 640x480 and stretch that to fit your display (a 1080p display for instance). It works quite fast, although it might stretch a 4:3 image to 16:9 aspect ratio depending on the person's native display. I use it to run games using ```--windowed --resolution 320x240``` using CEA-1, so I can play games at their native resolution.

gizmo98, 2016-06-07:
@joolswills I would like to switch to gliden64. I think we had this discussion some time ago. If we force CEA-1 we exclude analog output users. We also discussed the dispmanx sdl2 solution some time ago. You told me you would prefer a GLES stretch viewport/window option over a dispmanx option. Gliden64 is definitely the best plugin but I don't know what would be the best solution to force low res for all possible video outputs.

loganmc10, 2016-06-07:
@gizmo98 You can read here for a bit of a discussion on dispmanx scaling: https://www.raspberrypi.org/forums/viewtopic.php?f=33&t=7305

According to them, it is essentially "free" to scale the video using dispmanx, whereas using GLES would require from GPU time I would think.

Also, RetroArch is using dispmanx to accomplish this exact thing:
https://github.com/libretro/RetroArch/blob/master/gfx/drivers_context/vc_egl_ctx.c#L120-L145

joolswills, 2016-06-07:
I'm happy for the switch to gliden64 by default.

I can't remember the discussion regarding GLES, but perhaps that was so it was not RPI specific. If the SDL2 solution doesn't break anything else we could include that, but it won't fix it for the Mali SDL2 backend for example, which is why a fix in mupen64 itself would be best imho.

joolswills, 2016-06-07:
@loganmc10 @gizmo98 is a developer with experience of dispmanx - he did that portion of code you are references in RetroArch

dankcushions, 2016-06-07:
I really like the resolution SDL fix, as my (and others?) LCD TV has really horrible internal scaling from CEA1 (640x480) to 1080p, plus the performance improvements. That said, what annoys me about dispmanx scaling is that it always applies a very blurry bilinear filter. There must be ways to get around this as not everything from a low-res source using it to scale to native res has this 'problem', but i don't know how. some talk on this here: https://github.com/raspberrypi/firmware/issues/104

i would also suggest we use the 3-point filtering by default with GlideN64 -it's in the mupen64.cfg, but it defaults to standard bilinear filtering, which isn't authentic, and perhaps performs slower(?). However the glide section of the mupen cfg is empty until you run a game, so I don't know how you'd script that.

not sure about GPU split stuff - glide's texture cache might be huge but it can't be necessary. i've never had issues at the default 256MB split. in fact, you'll more likely get issues with other things that need lots of system ram (building mame from source, etc), if you raise it.

gles2n64 is better for mario kart 64, mario 64, f-zero and a bunch of other 'basic' games. it can run these at 1080p on a pi2 without issue. GLideN64 is more consistent (and accurate?), of course, but the frame rates and resolutions are much lower on these games. i don't see the point of the other plugins but i'm sure there's some usecase :)

gizmo98, 2016-06-07:
@loganmc10 
It should be tested if your patch stretches 640x480 to 1980x1080. If your patch keeps the source aspect ratio everything should be fine.

loganmc10, 2016-06-07:
@dankcushions To set defaults in mupen64plus.cfg you just need to put a stripped down version of mupen64plus.cfg in the place where it is going to get written, and make sure the section has the proper "Version" in it, so to set the filtering mode to bilinear by default you just need to create a mupen64plus.cfg like this:

```
[Video-GLideN64]
configVersion = 10
bilinearMode = False
```
Although configVersion might be different for you depending what version of GLideN64 you are using

loganmc10, 2016-06-07:
@gizmo98 Ya I should do that, I assume it doesn't keep the source aspect ratio, I'll probably need to add something like this:
https://github.com/libretro/RetroArch/blob/master/gfx/drivers_context/vc_egl_ctx.c#L132-L139

dankcushions, 2016-06-07:
@loganmc10 
great! I'll try that later and do a PR if it works.

I might also do the same to lower the default texturecache to 192 (pi2/3 default VRAM * 3/4, as per https://github.com/mupen64plus-ae/mupen64plus-ae/blob/master/jni/mupen64plus-video-gliden64/src/mupenplus/Config_mupenplus.cpp#L62) if that makes sense as an interim.

dankcushions, 2016-06-07:
@joolswills is there an iniFunc/etc to add a line to a cfg like `[Video-GLideN64]`, as in, with no parameters? I think I need to do that for these settings (which I am able to add to the cfg with iniSet) to take. cheers!

joolswills, 2016-06-09:
you can use "addLineToFile" or so, although that makes a backup, or just append manually with `echo "blah" >>"$file"`

dankcushions, 2016-06-09:
@joolswills I suppose it should to do something like the initSet function and only add the line if it doesn't already exist. that said, from what i've seen the mupen64plus install script deletes the old cfg every time you update... so maybe it's ok just to echo the line in. i'll test!

joolswills, 2016-06-09:
We probably don't want to delete the config though - that should be changed. you can do

```
if ! grep -q "[Video-GLideN64]" "$md_conf_root/n64/mupen64plus.cfg"; then
  echo "[Video-GLideN64]" >> "$md_conf_root/n64/mupen64plus.cfg"
fi
```
or so

joolswills, 2016-06-09:
What would be best would be to do like in retroarch.sh - if the config exists, we set up a mupen64plus.cfg.dpkg-dist, that way we never override users settings, but they can reference the default new config.

dankcushions, 2016-06-09:
Done! Tested a fresh install and an update, and it works as expected - install creates the section - updates create an .rp-dist of the new settings and doesn't overwrite. @joolswills, sorry, just re-read your comment - would you prefer it was dpkg-dist instead?

I've created the PR: https://github.com/RetroPie/RetroPie-Setup/pull/1518
does it need anything else? i'm not sure we should set GLideN64 as the default until we've settled on a resolution solution as the performance is terrible above CEA1.

joolswills, 2016-06-10:
rp-dist is correct - thanks.

closed 2016-06-10 by joolswills