RetroPie/RetroPie-Setup #1523: Can you add the TI-99/4A  (TI99SIM) to the RetroPi image  (here are the installation instructions):

globeron, 2016-06-12:
Dear developers,
(Video with install details): https://www.youtube.com/watch?v=3BT6eXlbO-Q
1-2-3 step Installation procedure:
http://atariage.com/forums/topic/250767-how-to-install-ti-994a-ti-99sim-on-retropie-v36-raspberry-pi2-or-pi3/#entry3491795

For 2x reasons would it be possible to include this in the RetroPi image? 
* I managed to get it installed, but now I get more questions about how to install the TI-99/4A on the RetroPi (see the AtariAge Forum)...another option is to create an .img, but that one is large (8 Gb SD)
* There is a large TI-99/4A community.

* TI99SIM is written by: ti99sim@mrousseau.org  (Marc Rousseau, maybe check with him first as well)

* (Note TI-99/4A ROMS, GROMS, SPEECH Binaries, should not be included, as well as Game roms, etc. these should be kept separately and the end-user still need to upload them)
(in my installation procedure I linked to other sources where you can find it, otherwise it will still be quite difficult to get it working).

HerbFargus, 2016-06-12:
Been there, done that.

https://github.com/RetroPie/RetroPie-Setup/issues/1435

closed 2016-06-12 by HerbFargus

globeron, 2016-06-12:
Thank for the fast reaction,  (it is not for myself) but my question is can it be included in the standard RetroPIE image? Then I can point TI-99/4A users on the AtariAge forum to download it from here (instead of each person going through the complete installation procedure)?    (= Effeciency, like the Nintendo's, Atari, etc.)

joolswills, 2016-06-12:
It is in experimental currently, and may move to optional, but will not be included by default in the image. The new modular retropie-setup script makes it easy to install additional stuff.

joolswills, 2016-06-12:
it just isn't popular enough to be included by default (sorry), and it doesn't have automatic controller setup - we try to keep the basic install to stuff that is simpler to use out of the box. It's only a few presses away to get it installed.

HerbFargus, 2016-06-12:
It's in the docs: https://github.com/retropie/retropie-setup/wiki/TI99 feel free to supplement the wiki page with any more relevant information as I am not too familiar with the ti99

globeron, 2016-06-12:
Hi Joolswills, 
thanks for the reply  (I am not an expert in RetroPie setup, etc., I am actually also not a GitHub user, but I had an account somehow). 

Re: Popularity - someone (cannot remember who) mentioned that on AtariAge the TI-99/4A was one of most active forums, I am not sure how to visualise the statistics.  Also on the broadcast "Floppydays", there were 5 dedicated episodes for this retro computer.

There is quite a lot of development ongoing over the last years, also hardware wise.  I am not sure what you mean with automatic controller setup (do you mean the PS/3 type of controllers?), I am using Oker types (see the video) and it worked out of the box with the TI-99/4A, I only had to add a USB-keyboard as key presses need to be done, but it works like a charm.   I also run TISIM on a Sony PSP  (so that it is portable to bring along) after installation, all keys, controllers, all worked.  That's what I want with the RetroPie, so it is easy to travel along with and connect it with HDMI to
an LCD TV.

Re: wiki pedia:  let me try to do the installation again with the latest RetroPie image and improve the description of the installation steps and put in there, I think it is good to have it online here. 
Also there is a very large group who can help here until bit/assembler levels (but that is not needed as far I can see).

PS. I am not sure if I get updates to my e-mail  (as I am not logged in to Github all the time)

HerbFargus, 2016-06-12:
Just a note on the wiki. In terms of formatting you can mention a note on it being in the experimental section of the setup script and then just has a link to the setup script on this page here which details installing individual packages:  https://github.com/RetroPie/RetroPie-Setup/wiki/Updating-RetroPie

And then if you have more advanced configurations like controller config info, emulator specific configs etc, those can be added at the bottom

joolswills, 2016-06-12:
we have a system that configures libretro emulators, and some others based on the users input. so the mapping is correct no matter the controller used. With the next retropie image, the amount of stuff pre-installed will be less - we want to keep our initial image small, so we have improved how additional packages are installed to make it easier for people to install/uninstall what they want - and then keep that selection up to date.

if you have the latest version of the retropie-setup script, it should be no more than a few button presses to get the emulator installed (and the time to get the bios files and put them in the right place)

globeron, 2016-06-12:
HerbFargus:  Wiki TI99 hope it is okay that  I have added the links at the bottom (for now) and will do the updates later.    

 (I have not seen the latest RetroPi image, but does it mean the base is there already (TI99SIM) and only the ROM (=Operating System) and ROM/GROM/Binaries/CTG files (= modules or called "Roms" / games, etc.) need to be installed?

PS. I still need to sort out the GUI (like Paradroyd is showing in his video to show a sample of the module to be loaded).

globeron, 2016-06-12:
Okay that sounds great if the installation is becoming easier!  The TI99SIM was in a way still straightforward and everything worked fine, like others as well VideoPAC, Nintendo, but I have been struggling with the Nintendo 64 (on a Raspberry PI 2+ B) as the sound is not so good  (jitter/latency) but the graphics are still okay (like Mario Kart 64), was not sure if this is the software or the hardware (and went through several forums to start to do some tuning)

globeron, 2016-06-12:
PS, there are many TI-99 emulators in the market (I have them all on my TI99VIDEOS channel on youtube), there is one online js99er.net as well. 

But this one is popular, because it is easy to install:  http://www.harmlesslion.com/zips/classic99.zip   (and very easy to install,  unzip it and within 1 minute you run it already (with some sample modules built-in,  the author has the rights for the console-roms btw).   

It would be great to have an easy installation like the one above, but not sure if that is possible in Linux.

globeron, 2016-06-26:
Hi for the new TI-99/Sim 0.14.0 (arm release 25 June 2016) and RetroPie v3.8.1 (4 June 2016), I made a more detailed installation procedure: http://www.globeron.com/freedownload/services/TI99/TI-99-Sim-Installation-on-RaspberryPI2ModelB-RetroPieV381-DocV2.pdf

but I have a few questions:
1. Audio on HDMI was not working (see the resolution in the PDF), by uncommenting the hdmi config  parameters,  still audio is not loud enough and the TV need to be put very loud to hear it.

2. Somehow I cannot figure out how the gamelist.xml gets updated (I tried manual, tried via Scraper), I removed it and it gets automatically generated, but by adding the .ctg files into the /home/pi/RetroPie/roms/ti99sim  directory is not adding the games in the gamelist.xml file.
.
3. Also I am trying to get the graphics inserted into the gamelist.xml 
    (see min. 1.14 onwards in Paradroyd video: 
https://www.youtube.com/watch?  v=sHJKwX318_g&feature=youtu.be)

4. Also always a keyboard need to be attached  (I cannot like with other systems), press Select and Start at the same time to go back to the main menu. 

Also it would be good if buttons could be programmed (e.g. keystrokes like 1,2,3, FCTN-8, FCTN-9, FCTN-=) as these are used mostly in TI-99/4A games  (in Windows we can do it using JoytoKey to configure any keyboard button to the controller: http://joytokey.net/en/)