RetroPie/RetroPie-Setup #1525: Add Support for BlissBox 4-Play

stevetb, 2016-06-12:
The BlissBox 4-Play is a wired controller adapter which supports pretty much every wired controller known to the emulation community.  Currently I believe the device is supported on Windows and Linux based operating sytems, possibly Mac as well.

**Here is a FAQ from their site about details:**
http://skunx7.wix.com/bliss-box#!home/mainPage

I wrote a post on their forum and it appears that support did exist to some extent on RetroPie but it was never 100% supported.  
http://blissbox.freeforums.org/blissbox-4-play-retropie-3-7-not-working-t410.html

This is not a very popular device currently but this is a company which has been trying to grow and spread the word about their device to the emulation community.  As an owner of the device, I would say it works very well for Windows.  It's not cheap at $120, but it provides a lot of flexibility.  

This is the perfect device for the RetroPie Project.

ulao, 2016-06-13:
Hello, I'm the Bliss-Box 4-play developer and although I do not know much about RetroPie I figured this info will be of some help.

1) The 4-play consists of 4 identical HID's Each on it's own USB port. There is a generic HUB inside. 
2) Each of these 4 devices are HID based gamepads (no driver needed ).
3) As of the latest firmware (1.54) we have seen little issues with linux. Macs and windows Os's are working.
4) The Descriptor is large and recognized in Windows, Mac, most all Linux's ,and the ps3. My guess is this may be the root of the problem.

```
///// gamepad	
0x05,0x01,  //    Usage Page Generic Desktop
0x09,0x05,  //    USAGE (Game Pad)
0xA1,0x01,  //    Collection Application
0x85,0x01,  //     Report ID 1
		
	//button
	0x05, 0x09, // USAGE_PAGE (Button)
	0x19, 0x01, // USAGE_MINIMUM (Button 1)
	0x29, 0x18, // USAGE_MAXIMUM (Button 24)
	0x15, 0x00, // LOGICAL_MINIMUM (0)
	0x25, 0x01, // LOGICAL_MAXIMUM (1)
	0x75, 0x01, // REPORT_SIZE (1)
	0x95, 0x18, // REPORT_COUNT (24)
	0x81, 0x02, //  INPUT	

	
	0xa1, 0x00, //   COLLECTION (Physical) ( must be in a collection for open-emu )		
		//axis
		0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
		0x09, 0x30,                    //     USAGE (X)
		0x09, 0x31,                    //     USAGE (Y)
		0x09, 0x32,                    //     USAGE (Z)
		0x09, 0x33,                    //     USAGE (Rx)
		0x09, 0x34,                    //     USAGE (Ry)
		0x09, 0x35,                    //     USAGE (Rz)
		0x09, 0x36,                    //     USAGE slider
		0x09, 0x37,                    //     USAGE  dial
		//0x09, 0x38,                  //     wheel ( cant figure this out )
		0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
		0x26, 0xFF, 0x00,              //     LOGICAL_MAXIMUM (255)
		0x35, 0x00,                    //     Physical Minimum (0)
		0x46, 0xFF, 0x00,              //     Physical Minimum (255)
		0x75, 0x08,                    //     REPORT_SIZE (8)
		0x95, 0x08,                    //     REPORT_COUNT (8)
		0x81, 0x02,                    //     INPUT

	0xc0,//end pointer phys collection.	
				
	//hat
	0x05, 0x01, // USAGE_PAGE (Generic Desktop)
	0x25, 0x07, // LOGICAL_MAXIMUM (7)
	0x46, 0x3b, 0x01, // PHYSICAL_MAXIMUM (315)
	0x75, 0x04, // REPORT_SIZE (4)
	0x95, 0x01, // REPORT_COUNT (1)
	0x65, 0x14, // UNIT (Eng Rot:Angular Pos)
	0x09, 0x39, // USAGE (Hat switch)
	0x81, 0x42, // INPUT (Data,Var,Abs,Null)
	0x65, 0x00, // UNIT (None)
	0x95, 0x01, // REPORT_COUNT (1)
	0x81, 0x01, // INPUT (Cnst,Ary,Abs)	

	/**  to make home work the button needs to be the first int he poll, also this vendor specific needs to
	     be here, at least one pressure. I left it as 2, because any more is another 8 ms poll**/
		//ps3 - I think this is just for pressure... the other feature needs to be on report ID 0, so it has to go on the mouse.
	
//0xa1, 0x00,  //   COLLECTION (Physical) 
		//if this is not here the playstation does not set the ps3mode, so it must be here. 
		0x06, 0x00, 0xff, // USAGE_PAGE (Vendor Specific)
		0x09, 0x20, // Unknown
		0x09, 0x21, // Unknown
		//0x09, 0x22, // Unknown
[... 640 lines cut ...]
			0x29, 0x03,                    //     USAGE_MAXIMUM (Button 3)
			0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
			0x25, 0x01,                    //     LOGICAL_MAXIMUM (1)
			0x75, 0x01,                    //     REPORT_SIZE (1)
			0x95, 0x03,                    //     REPORT_COUNT (3)
			0x81, 0x02,                    //     INPUT (Data,Var,Abs)
			//padding
			0x75, 0x01,                    //     REPORT_SIZE (1)
			0x95, 0x05,                    //     REPORT_COUNT (5)
			0x81, 0x03,                    //     INPUT (Cnst,Var,Abs)

			0x05, 0x01,                    //     USAGE_PAGE (Generic Desktop)
			0x09, 0x30,                    //     USAGE (X)
			0x09, 0x31,                    //     USAGE (Y)
			//0x09, 0x38,                    //     USAGE (Wheel)
			0x15, 0x00,                    //     LOGICAL_MINIMUM (0)
			0x25, 0x00, 0x04,              //     LOGICAL_MAXIMUM (1k)
			0x75, 0x08,                    //     REPORT_SIZE (8)
			0x95, 0x02,                    //     REPORT_COUNT (2)
			0x81, 0x06,                    //     INPUT (Data,Var,Rel)			
			
		0xC0,                          //   END_COLLECTION
    0xC0                          // END COLLECTION

```

joolswills, 2016-06-13:
According to http://blissbox.freeforums.org/blissbox-4-play-on-raspberry-pi-how-to-t398.html the problem is linked to Emulation Station and SDL2 - if SDL2 doesn't recognise it a a joystick it needs to be fixed there (We have a slightly custom SDL2 library, but ideally fixes go upstream).

Perhaps it can be tested with something like https://github.com/Grumbel/sdl-jstest - I believe the sdl2 sources have a joystick test also.

joolswills, 2016-06-13:
Some interesting results when googling the device ;-)

biscuits99, 2016-06-13:
Just a note to say I have one of these, and tried for about an hour to get it to work in RetroPie (resulting post http://blissbox.freeforums.org/post2424.html?p2424) and failed - jstest saw it and detected movement, but Retroarch didnt play at all - even though they have a udev file here:
https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Bliss-Box-4play.cfg

I love the idea of the blissbox for a retro setup, and am happy to test various configs if that helps anyone (or even post my unit off to someone who is keen to try testing).

stevetb, 2016-06-13:
Perhaps we could get the developer to donate a few to the project in exchange for RetroPie integration.  I think once support is added, he likely would see a spike in sales for his device.  The community really would dig this little device I think.

stevetb, 2016-06-13:
@joolswills yeah, I laughed about the name too.  Especially with a company called "Bliss Box".

ulao, 2016-12-14:
The 4-play has always worked with the pi you just need to understand linux errors. The problem you are having is fixed with a simple power adapter. Unfortunately the shipping, cost, and need for this is an overkill. Though, an easy item to buy, See this post for details.
http://blissbox.freeforums.org/post2871.html#p2871

biscuits99, 2016-12-14:
@ulao I'm getting "The requested topic does not exist." on that link - can you confirm the address?

ulao, 2016-12-14:
Sorry, http://blissbox.freeforums.org/blissbox-4-play-on-raspberry-pi-how-to-t406.html

ulao, 2016-12-14:
Looks like the link got messed up.http://blissbox.freeforums.org/blissbox-4-play-on-raspberry-pi-how-to-t406.html

thx

ulao, 2016-12-18:
A new post has been made, this one allows the 4-play to work in sdl2 mode so that ES and RA both work correctly.  http://blissbox.freeforums.org/post2919.html#p2919