RetroPie/RetroPie-Setup #1545: Seperate out TurboGrafx CD

stevetb, 2016-06-24:
All:

Following on Reddit the community would like to have the TurboGrafx 16 and the TurboGrafx CD games separated.  Currently they reside in the same roms folder.  This is a very similar situation to Sega Genesis and Sega CD, so it would make sense to me to separate.

People are finding ways to make it work but an official  #separation would be great.

**Here is one of the threads:**
https://www.reddit.com/r/RetroPie/comments/4oof4s/possible_to_seperate_turbografxcd_games_from_tg16/

Thank you!
-Steve

stevetb, 2016-06-24:
Here is another thread with images:
https://www.reddit.com/r/RetroPie/comments/4owsx4/i_figured_it_out_tgcd_has_its_own_folder/

stevetb, 2016-06-24:
And......another thread:
https://www.reddit.com/r/RetroPie/comments/4oof4s/possible_to_seperate_turbografxcd_games_from_tg16/

joolswills, 2016-06-24:
I'm not currently convinced of this - it has come up before and I wasn't for the change - more work for themers, and we already have so many systems. However some feedback from the retropie forum users would be useful, and if most people want it, we could look into it. I'd be interested in the view of @HerbFargus also.

HerbFargus, 2016-06-24:
How many more systems do we want to split? There are also only 21 CD games  44 if you count super CD. Perhaps if emulationstation loaded systems dynamically but the more systems we add the more maintainence for themes and the more white screen issues on themes with unique wallpapers. Idk I'm not really sold on it.

stevetb, 2016-06-24:
Looking at the Redump .DAT there are 320 disc images:  311 Japanese / 9 USA / 0 European.  So not a very popular system in the USA for sure, but it had a very strong following in Japan.  The only reason I bring it up is to offer the users that want it the ability to break them out.  Most users are probably not even aware NEC had a TurboGrafx CD peripheral.  The only other system I can think of which has this dilemma is Sega Genesis / Sega CD / 32X.

I'm not heartbroken if this doesn't happen, just personally liked the idea from a user perspective.

Thank you both.

joolswills, 2016-06-24:
Where do we draw the line ? eg vic 20, c128, pet, plus 4, vic 20 instead of just current c64 folder (which we use for vice which does all those systems), spectrum 48k, spectrum 128k+, +3, oric atmos oric-1, telestrat, pravetz,  amiga 500, amiga 1200, cd32, 4 models of trs-80, aatri st, atari ste, 

I'm not against splitting more out, if a majority of people want it, but perhaps it should be optional rather than by default so themers don't have a hard time. We also need to add the extra systems to the internal db in ES.

joolswills, 2016-06-24:
I should have said - I'm not against splitting more out in general, but im not sold on this change specifically - at least right now. But it's something that could be discussed further.

MetalManTN, 2016-06-27:
I would like to vote in favor of splitting out the Turbo CD. I certainly understand the arguments against doing this, but the Turbo and Turbo CD are my personal favorite system of all time and therefore I am biased. I will also say that the Turbo CD was the first CD console to be released, so I believe that makes it somewhat special and deserving of its own system entry. Again, I understand the arguments against but I do hope this system receives further consideration. Thanks.

HerbFargus, 2016-06-27:
It still needs discussion and before its even included it needs to be added to the default carbon theme. but in the mean time I have a branch with the changes if we ever do it (so if you want to set the changes manually you can see [here](https://github.com/HerbFargus/RetroPie-Setup/commit/59ad026e19c475a9d52b731dcf0036a0099c8d9f))

mxstar1, 2016-06-27:
This is actually very easy to do manually. I have split PCengine into 5 different themes/systems so now i can select pc engine, pc engine cd, turbografx 16, turbografx 16 cd, or supergrafx from emulationstation. This is probably a dirty way to do it, but its simple and easy. Since they all use the same emulators, you just create more pc engine entries in the es_systems.cfg, but name them according to the system you want to show. then change the rom folder path to individual folders for each system, and give each systema  different named theme. now to build the themes, just copy the theme for pc engine ( I used carbon because its xml points to generic named files, so it wont need any edditing.) now all you need to change is the system.svg. this is the hardest part. you need to find logos for the systems and convert them to SVG format. then drop them in each of your newly created theme folders in the correct place named system.svg. thats it, load up those new rom folders and it works like a charm, supergrafx you will need to change the emulator on a per rom basis, because if you change the emulator per system, it will change all 5 of these, because they are essentially the same system, just using different themes and rom folders.

mxstar1, 2016-06-27:
if you want, i can give you my custom made themes for carbon for these systems to make that part easier for you. Dev's can have them too if they want to incorporate them into carbon. just let me know

craigerskine, 2016-06-28:
I vote for the split.

I even made a TurboDuo SVG that would be free to use. [turbo-duo.svg.zip](https://github.com/RetroPie/RetroPie-Setup/files/337509/turbo-duo.svg.zip)

I use this in my custom ES theme because I mod my systems.cfg file to force this TG and TG-CD split (among a dozen other logical changes).

joolswills, 2016-06-28:
There is no need for people to post that they vote for it - this is not a poll. Platforms like the Vic 20 are arguably more valid for splitting than this - as it's a completely separate machine with different hardware from the c-64. There are many others too (see above).

This isn't going to happen now, but I might revisit it in the future.

stevetb, 2016-06-28:
@joolswills understood.  Thank you for listening.  We can revisit again in the future.

craigerskine, 2016-06-30:
Regardless, there is no difference between the separation of Genesis/Sega CD and TurboGrafx/Turbo Duo/CD. If Turbo is grouped then Sega CD should be lumped into Genesis/Megadrive.

1031982, 2016-08-25:
I just made it an option manually. It's not too difficult.

stevetb, 2016-08-25:
1031982:  Could you detail out the solution so we can test on our end as users?

1031982, 2016-08-25:
Sure, with version 4 of RetroPie (the most recent version), edit the /etc/emulationstation/es_systems.cgf file.
Add the following where you want the system to show up in the list : 
```
  <system>
    <name>pcecd</name>
    <fullname>TurboGrafx-CD (PC Engine CD-ROM)</fullname>
    <path>/home/pi/RetroPie/roms/pcecd</path>
    <extension>.pce .cue .zip .PCE .CUE .ZIP</extension>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ pcengine %ROM%</command>
    <platform>pcecd</platform>
    <theme>tg-cd</theme>
    <directlaunch/>
  </system>
```
Save the file, and then open the ROMS folder.
Make a new folder, all lowercase, called "pcecd" without quotes.
Put all the TGCD roms in there.
Now you have a designated TGCD section.
Make sure to backup the config file, as updating will overwrite it. 
You can make it so that it won't be overwritten by copying the config file to the /opt/retropie/configs/all/emulationstation folder.
I suggest making sure that everything works and you are happy with it BEFORE doing that.

stevetb, 2016-08-26:
@1031982 Thank you for the solution, it will work for some users for 4.0+.  I think most users would still like to see a native solution, just like we have for Sega CD.  In time I think it will come, its a historically important system.

1031982, 2016-08-26:
I would love to see it implemented natively. Fortunately, this system is included in the Carbon theme, so it looks like it's being planned.
In 3.x, I had to copy the PC Engine theme into a new folder to properly have it listed.

Darknior, 2017-02-03:
For my part i'm ok with @joolswills  👍

I always think i will separate the systems in my Video Game Museum, BUT i think also make it in ES source code is a bad idea. If not we must separate all the systems and it is impossible .... AMIGA 500 1200 CD ...
It is an SKIN separate problem and @1031982 give the good solution i already use for OpenBOR.

TheyCallMeCoco, 2017-02-14:
Now that the white screen of death issue is fixed in ES, could this be added?

stevetb, 2017-02-15:
@Darknior - You do understand this is no different then Sega Genesis / Sega CD / Sega 32x, right?  These three all have their own platform listed, which was the correct decision to split them all out.  We just want the same treatment for TurboGrafx CD.

joolswills, 2017-02-15:
@stevetb that decision was made a long time ago before we had so many systems. 

It's not just about the white screen of death, it's also about having a sensible number of systems (we already have so many), for themers etc.

TurboGrafx CD is just a cd addon - and if splitting out stuff, I will more likely split out things like vic-20 which are completely different systems.

I have gone into more detail earlier regarding my decision. You are free to split them out how you like yourselves in the meantime.

stevetb, 2017-02-16:
@joolswills - Would it be best at this point in time to just close this?  Otherwise I am sure the discussion on this will never end.

1031982, 2017-02-16:
@stevetb At this point, I wouldn't be surprised to see the Sega Genesis / Sega CD / Sega 32X merged into one system listing.
At the minimal, the Genesis and CD being merged makes sense as most of the games only needed the 3-button controller. 

Now, overall I appreciate the fact there is an easy option for adding the TG CD where you only need to add to the systems list. It's much easier then having to add folders and make additional theme artwork.
If the Genesis line was to be merged, I would prefer that it's done in the same way so that the systems can simply be added to the systems list.
The white screen of death is a problem for those who don't have any experience with Linux. The point of RetroPie is to be an easy setup to for running emulations of owned classic games. To eliminate problems combining systems makes sense. For enthusiasts, purists, etc, the RetroPie team has made it fairly easy to add additional systems. Anyone willing to change these things is well aware of the risks, so it would seem the middle ground is rather well done IMO.
I have gone as far as to re-orginise the systems order to put them in company and chronological order.
All the Nintendo systems are in order one after the other in order of release date. The last thing I have in the list is the RetroPie settings.
If I have problems, the first thing I do is restore the original configuration files. Took me a few hours to get it how I wanted, and that's using the Carbon theme.
Attached is a GIF of me scrolling through the systems list so you all can see how my adjustments look.
I want to point out that I had to make numerous adjustments for this all to work on my Raspberry Pi 3, and ocasionally I get a problem where I must SSH into my Pi and either kill Emulation Station or simply reboot the whole PI.
[image]

joolswills, 2017-02-16:
I am happy to leave it open, as I may well add some gui functionality to handle this at some point. However, I would hope people read through before commenting, as posts like "Can we now do this" don't really help :)

Pundan, 2017-10-17:
Separate them! In my config I've separated the CD games and the SuperGrafx games from the other TurboGrafx/PC-Engine games. I've even gone as far as separating the Famicom and Super Famicom games from NES/SNES. Luckily for me, the theme I'm using (pixel-metadata) had support for all this.

I hate having a list of thousands of games, then you don't know if it's the Japanese version or not. But the completionist in me feels like if I don't have the Japanese PC-Engine games also, I might miss out so that's why I separate them.

c0d3h4x0r, 2018-11-21:
Another strong argument for separating them: to get TG16/PCE CD-ROM games working, you have to prep the files in a completely different way.  The only way I've found to get them working is to have cue/bin file pairs with the cue file listing the exact casing of the bin file.  Combining them into a .7z or .zip file prevents the emulator from working with them.  Given the special/non-standard treatment they require, it makes a lot of sense to separate them out.  Additionally, the TG16/PCE artwork in the themes I'm using don't depict the version of the system with the CD-ROM drive attached... so, for example, anyone who's going to argue that MegaDrive/MegaCD/32X should all be one entry should also be arguing for the artwork in all the themes to be updated to depict a complete MegaDrive+MegaCD+32X system, and similarly, anyone who's going to argue that PCE+PCECD should all be one entry should also be arguing for the artwork in all themes to be updated to depict a complete PCE+PCECD (or a Duo).  Right now, it's all just an inconsistent mess.

krcroft, 2019-03-05:
Another vote for splitting.

The CD hardware quadrupled the available GPU memory (and gave it CDDA sound) making it a far more capable system than the TG16. The memory allowed developers to better use the GPU and making well-developed CD games a real treat visually (and audibly) compared to the competing 16-bit consoles SNES and Genesis (which typically crush the TG16). 

Currently someone has to "sprinkle in" their CD games amung their HuCard titles - and then it's a hackstack search when the ES list comes up.  Also with some titles overlapping, you better remember which is the HuCard vs. CD game otherwise it's trial and error.

XenuIsWatching, 2021-05-05:
Another good reason for splitting is for scraping through screenscraper fr (when using a tool such as skyscraper). The turbografx CD (as well as other platforms such as the Neo Geo CD and Jaguar CD) have separate platform IDs than their respective base platform.