RetroPie/RetroPie-Setup #1548: Issues installing GemRB with new RetroPie script

biscuits99, 2016-06-24:
I'm having issues installing GemRB, when the install is complete I cant find the config, rom or emulator folder anywhere.
I could well be doing something wrong, but looking at the Wiki the folders should be there.

I'm doing this:
- Write RetroPie 3.8.1
- Let it reboot/resize then configure iBuffalo USB in ES
- Quit ES then run the retropie setup from a terminal session (as I thought maybe ES was causing issues when I tried via the GUI menu in ES)
- Update RetroPie setup script
- Install gemRB, which appears to complete successfully as it shows as 'installed' on the list

But then I cant find any config folder or rom folder etc..

You can see the process I follow here:
https://youtu.be/mjBRgZLsdVM

biscuits99, 2016-06-24:
Ignore that, I've been in an idiot. For ref its in the ports directory.

closed 2016-06-24 by biscuits99

joolswills, 2016-06-24:
Looking at your video, you didn't look in `/opt/retropie/configs/ports` or `~/RetroPie/ports`

The wiki configs location is wrong, as all ports configs are in the ports/ subfolder now. This was mentioned in a changelog.

joolswills, 2016-06-24:
When opening a ticket, it does say to use the forum first also :-)

biscuits99, 2016-06-24:
Yep, sorry. Wiki now updated.

HerbFargus, 2016-06-24:
Thanks @floob :)