RetroPie/RetroPie-Setup #1552: fix default gcc

HerbFargus, 2016-06-28:
breaks on my linux mint 17.3 ubuntu 14.04 build

mmatyas, 2016-06-28:
I can also confirm this on Ubuntu 16.04. Using GCC 4.9, the build fails with the following error:

```
[ 98%] Building CXX object es-app/CMakeFiles/emulationstation.dir/src/views/SystemView.cpp.o
[ 99%] Building CXX object es-app/CMakeFiles/emulationstation.dir/src/views/ViewController.cpp.o
[100%] Linking CXX executable ../emulationstation
CMakeFiles/emulationstation.dir/src/main.cpp.o: In function `main':
main.cpp:(.text.startup+0x76): undefined reference to `boost::locale::generator::generate(std::string const&) const'
../libes-core.a(ThemeData.cpp.o): In function `ThemeData::getThemeSets()':
ThemeData.cpp:(.text+0x11c5): undefined reference to `boost::filesystem::path_traits::dispatch(boost::filesystem::directory_entry const&, std::string&)'
ThemeData.cpp:(.text+0x1210): undefined reference to `boost::filesystem::path_traits::dispatch(boost::filesystem::directory_entry const&, std::string&)'
collect2: error: ld returned 1 exit status
es-app/CMakeFiles/emulationstation.dir/build.make:903: recipe for target 'emulationstation' failed
make[2]: *** [emulationstation] Error 1
CMakeFiles/Makefile2:274: recipe for target 'es-app/CMakeFiles/emulationstation.dir/all' failed
make[1]: *** [es-app/CMakeFiles/emulationstation.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

The build works fine with GCC 5.

joolswills, 2016-06-29:
This is now fixed - a slightly different issue as the one herb was referring to was due to gcc 4.9 not being available. we only force it on jessie now (to work around issues with people having an older gcc-4.7 somehow).

Not sure why it wouldn't link correctly on 4,9 - but to revert your gcc back to how it was you can do `sudo __default_gcc_version=5 ./retropie_set.sh`

commit 6b4e7ac0 (2016-06-29): don't force default gcc on ubuntu - fixes #1552
closed 2016-06-29 by joolswills