RetroPie/RetroPie-Setup #1553: Use lr-nestopia as the default NES emulator due to better input-lag

dankcushions, 2016-06-28:
Please see http://libretro.com/forums/showthread.php?t=5428&p=41746&viewfull=1#post41746 - until https://github.com/libretro/libretro-fceumm/issues/45 is resolved (and I'm not sure it will be given that lr-nestopia exists), lr-nestopia has measurably better input lag. For this reason I think we should adopt it as the 'main' NES emulator, with lr-fceumm as the 'opt' (currently it is the other way around). I am sure I've seen some users complain about input lag in the NES emulator, so this should help!

One caveat here - I understand that lr-fceumm runs faster than lr-nestopia, but is this measurable on any Pi systems? I think there was a reason we have lr-fceumm as the current default and it might be this. I have a Pi 3 and I get 60FPS on both cores. Any Pi 2/1/0 users who can check? If lr-fceumm runs better, we could put a platform check in the script.

joolswills, 2016-06-28:
I do think the people complaining about input lag area relatively small number - and in most cases is due to the screen (ie not in the right mode, or a bad screen). I personally do not notice any lag with fceumm, however some people might be more sensitive to it.

We definitely need more testing, but I am reluctant to switch based on 1 frame of extra lag - it's also easy enough for people to switch. As you know, changing defaults often causes problems :-) I would need further convincing - also in terms of compatibility. Some rpi0/1 tests would be useful though.

dankcushions, 2016-06-28:
Have you read the link? I trust the assessment given the detail they've gone into, so I think it's for sure better in these terms, but agreed - I'm personally going to change but we need to get rpi0/1 confirmation first.

joolswills, 2016-06-28:
I read it yes.

joolswills, 2016-06-28:
Hence specifically mentioning the 1 additional frame delay

Brunnis, 2016-06-28:
1 frame might not be much, but considering that RetroPie with fceumm and snes9x-next (before my fix that was just committed) has 7-8 frames (117-133 ms) of lag even before adding the display's input lag, I'd say that we need to do what we can to reduce it. Adding a display with 3-4 frames of input lag makes the setup as a whole approach 200 ms. The reason that I started looking into this from the beginning was the fact that those 200 ms (yeah, my TV is slow) made Super Mario World really hard to play.

However, I do see where you're coming from. Although I personally prefer Nestopia and have been using it for a long time, both on my Pi 2 and Pi 3, I definitely see how switching could cause issues. Unless someone can provide compelling info regarding compatibility and RPi 0/1 tests, it might be better to simply wait for a fix in fceumm.

LodanZark, 2016-06-28:
nestopia has better accuracy and less input lag yet fceumm it's faster and has better compatibility with non-intro, unlicensed and specially hacked roms. Personally nestopia as standard core might lag on Rpi1 plus some users might complaint about some games being unable to run...

dankcushions, 2016-06-28:
the scripts can handle one default for one platform, one for another. that was my assumption - nestopia for 2/3, fceumm for 0/1. perhaps anyone who cares enough about the lag will read the wiki and do the change anyway...

Brunnis, 2016-06-29:
I've found a fix for the issue and created a pull request here: https://github.com/libretro/libretro-fceumm/pull/46

Talk about small fix (moving one line of code one line up...).

dankcushions, 2016-06-29:
even better! cheers @Brunnis :) I recommend we update the binaries once that one gets pulled. Closing.

closed 2016-06-29 by dankcushions

joolswills, 2016-06-29:
I have updated the binaries.