RetroPie/RetroPie-Setup #1555: PS3 Controller works in ES but not in games: case study.

metabolist, 2016-06-30:
This took me about 3 hours of my time so I thought I'd contribute by telling the story. Perhaps the conditions of my initial disaster and eventual solution could apply to you. Hail to all NoobS!

- System: RPi3, 32GB SD Card, + 5 controllers including 1 wireless keyboard (via dedicated USB1 dongle), 1 wireless mouse (via dedicated USB2 dongle), 1 SNES Joypad (USB wired), 1 PS3 Controller (wired most of the time), 1 XB360 Controller (wireless via dedicated XB360 dongle connected to USB). Note: I rarely use the XB360 Controller, so in most case the dedicated dongle is disconnected from the RPi3. However, all the drivers for these controllers are installed and have been configured multiple times with RetroArch.
- Retropie: 3.8.1.
- SD Card is full (300 MB free).

What happened:
- looking initially to update the configuration for PSX emulation in order to enhance display. I'm using a high end Asus monitor and the standard definition from PSX emulation was way too pixelated for my taste.
- I followed the instructions from a YouTube video here https://www.youtube.com/watch?v=r6-VOCyDoTc
- After the changes made manually in the config files, as per the video, I started a PSX game and opened RetroArch to adjust some settings, including for video and controllers.
- After the reboot, the PS3 controller (both digital and analog) would only work to browse ES. It wouldn't work in games or within the settings menus in RetroPie Setup.

What might have caused the problem:
- my assumption is that the problem occurred after I made the changes in RetroArch settings for the PS3 controller from digital to analog. It is only an assumption and I do not know if this is accurate.

What I tried / What made the situation worst:
- playing manually with config files for RetroArch and ES.: didn't work.
- deleting all the config files for the PSX emulation. I assumed, wrongly, that rebooting would have RetroPie re-generate all the files I deleted. I also started receiving error messages about PSX emulation. The final result was that no game would start at all under PSX emulation.
- After that fiasco, I decided to update all systems and packages from the option in RetroPie Setup. After this was done, PSX games would again launch but the PS3 Controller would still be useless in game and in settings menus.
- Deleted PS3 controller configs, deleted PS3 controller driver and reinstalling: didn't work.

Solution:
- I eventually saw that RetroPie would recognize at least 3 controllers whenever a game would start, regardless of the emulator - this shows in the texts in yellow before the game splashscreen.
- I also saw the PS3 controller was always recognized as a XB360 unit and would run under the xb360drvr driver.
- I deleted the xb360 driver - just disabling it wasn't sufficient.
- I paired the PS3 controller again and reset the UI settings for the controller.
- Working.

Aftermath / Lessons learned:
- don't play manually with the settings from the console / desktop (after you pressed F4 or ssh'ing). Instead, learn the menus in RetroPie Setup and use that to modify the settings there. I actually saw later that all the changes described in the YouTube video I mentioned above are available from the menus directly from RetroPie setup.
- don't connect more than 2 joysticks to RPi3.
- RetroPie shouldn't recognize any controller unless that controller is either connected or switched on. What's surprising is that RetroPie would always recognize the XB360 controller even though no dongle would be attached to RPi3 and XB360 was totally disconnected.

joolswills, 2016-07-03:
This is a bugtracker - please use the forum for howto/tips

closed 2016-07-03 by joolswills

joolswills, 2016-07-03:
(or documentation can be added to the wiki if you think you have a relevant change. thanks)