RetroPie/RetroPie-Setup #1562: Emulation station doesn't recognize wireless keyboard

virajmahesh, 2016-07-05:
I installed RetroPie on top of Raspbian by following the instructions for manual installation. However, when I try and launch emulation station, I get a message saying 'Welcome, No Gamepads Detected'. For some reason it doesn't respond to my wireless keyboard. Is there any way to fix this?

davidt99, 2016-07-09:
Have you pressed A on your keyboard? This message is normal when no gamepads are connected.

joolswills, 2016-07-09:
This should be posted to the forum as mentioned when opening an issue

>Please make sure you read the "guidelines for contributing" above before opening an issue. Support questions should go to the forum. Please post to the forum first, as many "issues" can be solved with configuration changes. Please remove this text before posting.

closed 2016-07-09 by joolswills