RetroPie/RetroPie-Setup #1570: Super Mario War - Controller mapping

silverkorn, 2016-07-12:
As in reference with issue #1409 and by personal experience, Super Mario War doesn't map the controller to the predefined keyboard keys neither auto-configure the controller(s) in it's settings.

joolswills, 2016-07-12:
Are you reporting a bug or is this a feature request ? Did you read though https://github.com/retropie/retropie-setup/wiki/super-mario-war ?

silverkorn, 2016-07-12:
I consider it as a bug since all emulators are mapping correctly.

If it's a feature request, it's a bit odd to force the end-users to get a keyboard there to setup their gamepads. As far as I could see, I always need to reselect the controller in the setup of the game each time I load it. :confused: 

Thank you.

joolswills, 2016-07-12:
It's not a bug. RetroPie automatically configures RetroArch, and a few other emulators, but we don't have the time to automatically configure every single application. SMW is not a high priority either, so unless someone contributes the auto configuration it isn't likely to happen. Also note that we are not responsible for the code for every application retropie can install.

joolswills, 2016-07-12:
Please use the forum first for posts like this as mentioned when you opened the issue

>Please make sure you read the "guidelines for contributing" above before opening an issue. Support questions should go to the forum. Please post to the forum first, as many "issues" can be solved with configuration changes. Please remove this text before posting.

Note: "Please post to the forum first,"

closed 2016-07-12 by joolswills