RetroPie/RetroPie-Setup #1575: Gamecon Driver - 4.0 beta2

deathly87, 2016-07-15:
Hi Guys,

i have a big problem to install the gamecon driver on the latest version.
At first i update the RetroPie-Setup script, then i want to install the gamecon over the driver package.
Everytime i get some issues. I have try to install the driver manually, but dont work. On 3.8.1 everything works.
 
Logs: http://pastebin.com/B9FuEeq4

BR
André 

PS: sorry for my bad english :P

joolswills, 2016-07-17:
You should post first on the forum as directed when opening an issue, and I will advise. I suspect you have run rpi-update to install a bleeding edge kernel - you will need to downgrade to the last packaged version for this.

joolswills, 2016-07-17:
if you haven't use rpi-update, you should make sure your kernel is up to date - `sudo apt-get update; sudo apt-get dist-upgrade`

closed 2016-07-17 by joolswills