RetroPie/RetroPie-Setup #1578: Cannot connect bluetooth keyboard

root42, 2016-07-18:
I have a BT keyboard I want to connect using the bluetooth setup menu. However the setup is searching forever. I noticed that the bluetooth service is set to non executable and is supposedly replaced by sixad. But that one does not seem to support keyboards...? The whole Bluetooth thing seems very broken to me. 

I use two wireless USB CSL gamepads and a BT keyboard on my Pi2 with BT dongle. How can I make this work? I have manually chmod a+x the bluetoothd and then it works fine -- until a reboot. it seems to me I do not need the sixad daemon.

root42, 2016-07-18:
More information: When turning my keyboard on, I can see these messages in the syslog:

`:1a:7d:da:71:10
Jul 18 09:12:45 the-shire kernel: [71792.840047] input: Broadcom Corp.   Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/bluetooth/hci0/hci0:69/0005:04E8:7021.000F/input/input14
Jul 18 09:12:45 the-shire kernel: [71792.841448] hid-generic 0005:04E8:7021.000F: input,hidraw0: BLUETOOTH HID v0.01 Keyboard [Broadcom Corp.   Keyboard] on 00`

joolswills, 2016-07-18:
You should post to the forum for support question as mentionedwhen opening an issue

>Please make sure you read the "guidelines for contributing" above before opening an issue. Support questions should go to the forum. Please post to the forum first, as many "issues" can be solved with configuration changes. Please remove this text before posting.

sixad is incompatible with the built in bluetooth stack. So if using ps3 cotnrollers you can't currently use a bluetooth keyboard as well. Hopefully this will change when we have more success pairing ps3 controllers without sixpair/sixad

closed 2016-07-18 by joolswills