RetroPie/RetroPie-Setup #1582: Add Option to Switch Region Specific Console Logos/Names

shirizaan, 2016-07-19:
Please add an option to the RetroPie-Setup script to switch consoles to what is appropriate for a specific region (i.e. Megadrive -> Genesis and PC Engine -> Turbografx 16). This can currently be done by editing the config files directly but is far from user friendly. It would also open up the possibility of other regions such as Super Nintendo -> Super Famicom, etc.

joolswills, 2016-07-20:
May consider this in a later release, but for now it's a low priority.

ProxyCell, 2016-07-22:
I would like to add one note that I feel is important to this:
Changing the name/logo will not re-order the platforms. So the TurboGrafx16 will still be alphabetically sorted as PC-Engine in the Emulation Station menu and not be near SNES but be closer to Ports still.
You may find it more convenient to leave them.

dukesilverfan, 2016-07-26:
ProcyCell, the current way to rename and change the logo is editing the es_systems.cfg file, and by re-ordering the file, you can re-order how the systems appear in Emulation Station. If joolswills decides to do this, I am sure that can be taken into account as well. It seems easy enough to follow the wiki and do this manually though, so I don't know how much benefit this enhancement offers.

joolswills, 2016-11-21:
https://github.com/RetroPie/RetroPie-Setup/pull/1759 deals with this

commit ccd51513 (2016-11-22): platforms.cfg / helpers - allow overriding of theme per system - fixes #1582
 * configs/all/platforms.cfg can now contain system_theme="name" to override a theme for a system. eg adding megadrive_theme="genesis" to configs/all/platforms.cfg will make sure emulationstation always uses the genesis theme for megadrive for our North American users.
 * simplify some of the names in platforms.cfg
closed 2016-11-22 by joolswills