RetroPie/RetroPie-Setup #1583: lr-gpSP segfaults

Pitel, 2016-07-20:
```
$ /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-gpsp/gpsp_libretro.so --config /opt/retropie/configs/gba/retroarch.cfg "/home/pi/RetroPie/roms/gba/1986 - Pokemon Emerald (U)(TrashMan).gba"
romtitle     : Pokemon Emerald (E/U)
gamepak title: POKEMON EMER
gamepak code : BPEE
gamepak maker: 01
INPUT gamepak title: POKEMON EMER
INPUT gamepak code : BPEE
INPUT gamepak maker: 01
found entry in over ini file.
[1]    7484 segmentation fault  /opt/retropie/emulators/retroarch/bin/retroarch -L  --config
```

Using latest updated RetroPie.

lr-gpSP was the default whn I first launched GBA game, and it crashed. Switching to lr-mgba "solved" the crash, and maybe it should be the default option. It has better performance than lr-vba-next on my Pi2.

dankcushions, 2016-07-20:
know issue: https://github.com/libretro/gpsp/issues/5

lr-gpsp is the fastest but maybe lr-mgba should be default for pi2 and up if it is always full speed on them? (I haven't tested this)

Pitel, 2016-07-20:
I might try to run [some demos](http://www.pouet.net/prodlist.php?order=views&platform%5B%5D=Gameboy+Advance) later, and report any noticable performance issue. Or is there any known game to test the speed of emulation?

Pitel, 2016-07-20:
Ok, I'm back with results. Worst I got from mgba was around 40 fps, and it was in scenes clearly not suitable for GBA (antialiased 3D rendering with physics simulation, etc.). gPSP on the other hand, rendered some of those demos completely wrong. Some "sprites" were not transparent (black, instead). So I'm gonna stay with mgba on my installation.

joolswills, 2016-07-20:
I have made mgba default for armv7 platforms now. I think this issue can be closed here though as it's not a retropie problem.

closed 2016-07-21 by joolswills