RetroPie/RetroPie-Setup #1585: [RPI2] directories ever created after RP update

nemo93, 2016-07-21:
I like to keep things tidy and clean whenever possible. Therefore I removed any directory in the roms/configs locations I'm not using. I end up with something like 13 directories (snes, nes, megadrive, etc). Though anytime I update RP via the script (item Update Retropie-Setup script), the 2 following directories keep from being created under configs:

* dreamcast
* ports

I have to delete those after the update. Please note that of course in case of a full update of RP (Manage >> Update all packages) more directories come back (all ?? need to check that) and have to be removed.

Not a big issue obviously :)

joolswills, 2016-07-21:
Did you post to the forum first ?

joolswills, 2016-07-21:
(I don't think this is a bug, and we request issues are open on the forum first - directories are only re-created for emulators that are intalled with retropie-setup 4.x - not sure what version you are running as you give no details. If you dont want dreamcast rom folder recreated remove reicast emulator)

closed 2016-07-21 by joolswills