RetroPie/RetroPie-Setup #1590: retropie.cfg not working with lr-snes9x-next

psthompson, 2016-07-23:
In 3.8.1, when using lr-snes9x-next as the default SNES emulator, the global and emulator specific retropie.cfg files do not work. When loading a rom, text displays that says Logitech F310 Not Configured. This may be an issue specific to the F310, but I do not have any other gamepads to test. The F310 also works with other emulators without any issue.

If you switch the default emulator for SNES to snes9x and use the /opt/retropie/emulators/pisnes/snes9x.cfg file, the gamepad works without issue.

psthompson, 2016-07-23:
It seems that this was being caused by configs that I had imported from an old version. I started fresh, and not everything works great!

closed 2016-07-23 by psthompson

joolswills, 2016-07-23:
Also please note the text when opening a ticket - this issue tracker is not used for support and even issue should be posted first to the forum.