RetroPie/RetroPie-Setup #1593: Can not build lr-mupen64plus from source

ProxyCell, 2016-07-25:
I'm using the latest retropie scripts from today and I just wanted to test building everything again, tried building lr-mupen64plus from the packages menu and this is being throw into the log:

> git clone --depth 1 --branch rpi2_fix "https://github.com/gizmo98/mupen64plus-libretro.git" "/home/pi/RetroPie-Setup/tmp/build/lr-mupen64plus"
> Cloning into '/home/pi/RetroPie-Setup/tmp/build/lr-mupen64plus'...
> warning: Could not find remote branch rpi2_fix to clone.
> fatal: Remote branch rpi2_fix not found in upstream origin
> /home/pi/RetroPie-Setup

joolswills, 2016-07-25:
@ProxyCell looks like the branch has been deleted. @gizmo98 is this branch no longer needed - should we switch back to upstream repo for lr-mupen64plus or ?

joolswills, 2016-07-25:
I have  switched to the upstream repository - at least for now.

gizmo98, 2016-07-25:
Oops. Accidental removal. ;-)

joolswills, 2016-07-25:
the upstream repo seems to work for me (tried a few video plugins)  - I remember there was an issue before why we didn't use it, but can't remember what ?

gizmo98, 2016-07-25:
gln64 was broken and it couldn't be build with neon opts.

joolswills, 2016-07-25:
if that is still the case we can go back to your branch if you have it locally still (and can re-push). - I guess ideally it would be fixed upstream though if you know what the issue is ?

gizmo98, 2016-07-25:
Could be loganmc10 has fixed the neon bug:
https://github.com/libretro/mupen64plus-libretro/commit/7fe29be8f2b02469bdccd3daf213dba4cbcc5411

gizmo98, 2016-07-25:
We can keep it as it is. We could also move lr-mupen64plus into the experimental section because compared to the standalone version it is neither performant nor compatible.