RetroPie/RetroPie-Setup #1598: Unplug f310 controller

thorst, 2016-07-27:
https://github.com/RetroPie/RetroPie-Setup/issues/912

I have the same issue as #912 with 4.0 rc1. Once i plug it in it will work for the duration of the game, but each time i enter a game it requires me to unplug and plug the controller back in.

In a similar vein, when I press select and X to enter the menu, i cannot navigate or select. If i plug in a keyboard and unplug the controller it will allow me to navigate and select or change options.

I have a pi3 and only have this one controller plugged into usb, no other usb devices attached.

joolswills, 2016-07-27:
Make sure the configs match here - you need to get the shoulder inputs right, or it messes with other button presses.

See https://github.com/retropie/retropie-setup/wiki/Logitech-controllers

joolswills, 2016-07-27:
I suspect your issue is related 

>The "left top" (L2) and "right top" (R2) triggers can be tricky to get right. If you press and release the trigger quickly, you may configure the negative (outward) axis, which is wrong. When this occurs, RetroArch emulators will need the L2/R2 buttons to be pressed once before registering any other input. If you see "left top" as -2 and/or "right top" as -5 then the axis is configured incorrectly.

thorst, 2016-07-27:
I actually think it has to do with this. I still need to test it out, but preliminarily it looks like the solution.
https://www.reddit.com/r/RetroPie/comments/3yh7o3/pi_zero_latest_retropie_logitech_f310/

> 
> **Herb_Fargus**
> 
> Tap the top right trigger to make it responsive. It's a weird firmware issue
> 
> 
> **ide_cdrom**
> 
> Woah, that was it! Wierd! Thanks! When you said it's a weird firmware issue, do you mean it's some oddity with RetroPie itself?
> 
> 
> **Herb_Fargus**
> 
> No rather some oddity with raspbian or rather the firmware that is used for the raspberry pi hardware itself. In theory you can downgrade your kernel and it should resolve it but I don't find tapping a button inconvenient enough to downgrade
>

thorst, 2016-07-27:
Yes, it seems if i press what used to be called R1 on the ps1 controller it will reconnect

closed 2016-07-27 by thorst