RetroPie/RetroPie-Setup #1599: RetroPie Menu does not show as installed in core packages after upgrading

movisman, 2016-07-27:
A purely academic cosmetic issue when updating the other night:
When navigating the RetroPie-Setup script, the 'RetroPie Menu' does not show as 'installed' in core packages after performing an upgrade.

Steps carried out to produce the issue:

- Was on v3.7
- Updated setup script to 4.0rc1, ran Quick Install and Update All Installed Packages
- checking out new UI, I can see which emulators and packages are installed as expected
- noticed the "RetroPie Menu" did not say 'installed' even though it is actually there and working fine
- removing and reinstalling package, now shows as 'installed' as expected.

Can the code be tweaked so that after an upgrade it shows the RetroPie Menu as 'installed'?

Cheers

joolswills, 2016-07-27:
Do you have a log - it should run an update hook to fix a few packages that don't show as installed.

joolswills, 2016-07-27:
(gzipped logs in ~/RetroPie/logs)

joolswills, 2016-07-27:
I can reproduce it (I had thought I had already sorted this - but perhaps from only 3.8 - it's slightly tricky due to the way the script is reloaded differently on each version after an update

joolswills, 2016-07-27:
actually it's fine here going from 3.7 to 4.0 I just forgot to remove a file (a VERSION file is generated in /opt/retropie which is used to tell what version we are coming from or if it doesn't exist on an older version the update hooks are run). I will try with a fresh 3.7 image, but so far I can't reproduce it. You didn't have the /opt/retropie/VERSION file somehow on the 3.7 install did you ?

joolswills, 2016-07-27:
I just installed a fresh 3.7 images on an sdcard - and did an "update retropie-setup script" - it fetched the latest script, relaunched it, and because /opt/retropie/VERSION was missing  - ran the update hooks. I then went to Packages / Core Packages, and retropie menu shows as installed correctly.

joolswills, 2016-07-27:
Your log would show me whether the hooks got run correctly, and if they did, the only thing it could be is that you didn't have a gamelist installed for the retropie menu which it uses to detect if the retropie menu is on the system.

movisman, 2016-07-27:
Hi,

Yeah i'm pretty sure I was on 3.7, I upgraded to that from 3.6 and 3.6 was the original image I started with. However my 3.7 instance prior to updating to 4.0rc1 would have had a slightly updated setup script post release. I'm certain I would have had a gamelist.xml for the RetroPie menu on the existing install as this displayed with issue?

I am not at home this eve, but when I am i'll extract the log for you?

Cheers

joolswills, 2016-07-27:
you can - although if you had updated to a version after 3.7 there may be no fix in your case (as I made a few different changes to the way the update mechanism works)

movisman, 2016-07-28:
Ah I see, yeah my system was a 3.6 binary based install from scratch, later updated to 3.7 (full binary install on top). After this I would have updated my RetroPie-Setup script at some point between 3.7 and 3.8 to get some of the commits. Then i've gone straight to 4.0rc1.

That's what I remember anyway :)

It's such a minor thing, figured I would report it anyway to try and help with general tidy up in case there was an issue in the code. Probably only affected a handful of users! If you still want the log let me know and i'll generate later.

joolswills, 2016-07-28:
Without being able to reproduce it, I don't think there is much I can do. What is odd, is that a quick install will install the retropie menu also, so I'm not sure how it didn't show up after that. If you can heklp me reproduce the issue, I can dig further or we can just not worry about it :-)

movisman, 2016-07-28:
Yeah it's weird, because the retropie menu was always there and functional from the ES menu, it just didn't flag it as 'installed' in the script after updating to 4.0rc1 from 3.x.

So it was purely cosmetic. All other packages showed as 'installed', just for some reason the script wasn't displaying 'installed' for this particular package - even though it was working fine. It must have had a gamelist and all associated images otherwise it wouldn't have worked before. Reinstalling the package sorted the display issue with the script not showing it as installed.

I have had a look at the logs and I personally cannot see anything strange, looking for 'retropiemenu' just shows an install as usual as part of the upgrade. It doesn't look there is any logging for removing packages.

I am thinking, the only way I can reproduce the issue now, is to take an image of today's config, restore my last backup from when I was on the older version, try to update again and see if it reoccurs. Once reproduced I can restore my current image. What info do you A) need from the older version and B) from the newer version post-update to look at it further?

Otherwise, happy to just not worry about it. It really is the least important of issues and would go unnoticed by most. Worst case, if other people experience it, is they think "oh the menu didn't install properly" and they remove and re-add it, which then makes it display as installed. Or they just ignore it because it still works.

joolswills, 2016-07-28:
I don't mind looking at it if you want to spend the time to reproduce - from the old version I would ideally need an image or a copy of everything in `/opt/retropie` and `/home/pi/RetroPie-Setup`

otherwise we can not worry about it as I think it is not a common case (I updated from a 3.7 and 3.8 image and both seemed to be ok)

movisman, 2016-07-29:
I was going to leave it, but as usual got curious. I was able to replicate. Here's how:

---
Reverted to 3.7 backup, confirmed there is no 'version' file in /opt/retropie

Update setup script from 3.7 to 4.0rc1
Script shows retropie menu as 'installed'

Run a quick install to perform update of the core packages
Once complete, script still shows retropie menu as 'installed'

All good at this point.

Exit the script to go back to ES
Go back to Retropie-Setup from the retropie menu

Script no longer shows retropie menu as 'installed'

Performing reboot of system, it still does not show retropie menu as 'installed'.
Retropie menu however functions fine. Gamelist and images all in place as expected.
---

I have the files you wanted from 3.7 but i'm not sure if they are of use in light of the above.

It looks like once the setup script is relaunched from ES after the update, it clears the 'installed' flag for the retropie menu. Why that happens i'm not sure.

Not a big deal clearly as it's cosmetic, hardly worth talking about tbh, but just passing on any bugs/strangeness I do experience as it might help with polishing the new script (which is excellent btw, way more user friendly for package management).

movisman, 2016-07-29:
In addition, when you click 'retropiemenu' within the package manager, I can only install from binary even though it is actually installed. So it isn't just the wording that's missing, the script genuinely thinks the package is not there.

movisman, 2016-07-31:
Just randomly spotted on Herb's page for updating RetroPie here:
https://github.com/RetroPie/RetroPie-Setup/wiki/Updating-RetroPie

Doesn't show the retropie menu as installed either on the screenshot post-update. Possible coincidence of course!!

joolswills, 2016-07-31:
They were made before I put the fix in to make it show up as installed.

joolswills, 2016-07-31:
I've not had time to look into this further, but will try from a 3.7 images following your instructions when I have some free time.

joolswills, 2016-07-31:
ok good news - I can reproduce the problem - I have no idea just yet why it gets removed - looks like it gets removed when exiting after the "quick install", but will debug further. Thanks.

movisman, 2016-07-31:
Great no worries! Cheers!

joolswills, 2016-07-31:
Please can you try it again as I believe I have worked it out. It was due to launching from the ES menu itself which launched calls a function in retropie-setup which calls the main retropie-setup menu, but when exiting would continue to run the old version, which I believe was messing with the retropiemenu install.

joolswills, 2016-07-31:
Whilst debugging this, I also made some other improvements to debugging/logging and error reporting, so thanks for bringing this to my attention.

joolswills, 2016-08-01:
Despite my change I'm still not certain it's fixed, and a test just failed, so I will continue later (I think I am on the right track anyway - just need to confirm the problem with some more debugging)

joolswills, 2016-08-01:
aah hmm. the problem is although I've fixed it for future issues like this, the old script doesn't have the fix of course, and that is what gets called. not sure of a workaround now, as the old script will always be called after - which will remove the folder. I think we might just need to document this upgrade issue.

The only possible fix I can think of is if the new script looks for older running scripts and kills them (eg the old one). That might be possible, but ugly.

joolswills, 2016-08-01:
Or it might be enough to put a file in the folder - as I think the old script wont remove it if it isn't empty. Ill try that.

joolswills, 2016-08-01:
I think I have  working fix in place now - as the older retropie 3.7 script removes the install folder because it's empty - now we create a file in it as a hack.

joolswills, 2016-08-01:
Have done a fresh test and all seems good.

movisman, 2016-08-01:
Brilliant, cheers for that.

Later tonight i'll reinstate the 3.7 backup once more, do another update and confirm that it's fixed for me also.

Cheers

movisman, 2016-08-01:
Hi Jools,

Just reinstated my old 3.7 backup for you and updated as per the above, and I can confirm the issue is fixed for me, so feel free to close.

Thanks again for the help, appreciate it.

joolswills, 2016-08-01:
Cheers.

commit e8649864 (2016-07-31): retropie-menu - when launching retropie-setup from menu - use exec so we don't return to an older running script after (which break things) - should fix #1599
closed 2016-08-01 by joolswills