RetroPie/RetroPie-Setup #1604: Additional uninstall option in setup script

movisman, 2016-08-01:
This is an enhancement request rather than an issue, so let me know if this is best posted on the forum.

When uninstalling emulators, I think it would be nice if there was an additional option to remove the emulator entirely including all configs, roms, gamelists, etc?

I like to keep my instance tidy, and since the move to the new script I enjoy the fact that everything is so much easier to manage from the GUI and it's nice and clear to the user what packages exist on the system.

Using the standard image as a example, once deployed there are around 10-15 emulators I will personally never use. I know I can just leave these in place with no ill-effects, but it's nice to take advantage of the new script functionality and get rid of them if desired to keep things streamlined. However, when I uninstall there is quite a lot of stuff left over, presumably so that if the emulator is reinstalled later, any configuration made, roms uploaded, gamelists created etc, would get reinstated for the user.

While I think that's the most sensible idea and certainly needs to be the default, do you think it's worth having an additional option in place, which gives the user the option for a more aggressive uninstall to clean things up?

This may also come in handy if a user is having issues with a particular emulator or has screwed up their config. With the option of a more aggressive uninstall, they would be able to wipe the emulator clean knowing everything has been removed. Then they can start afresh by reinstalling the emulator from binary.

Thoughts?

joolswills, 2016-08-01:
I intentionally left it out - although there is an complete uninstall of retropie. If you need a minimum set of things, you could just set it up over a raspbian lite image and install just what you need initially.

It is left out for various reasons - people accidentally removing their data, but also, currently there isn't tracking of all configs for everything, so not all modules know  what belongs to them. 

There should not be "a lot of stuff" left over though. Just the roms folders, and the configs in /opt/retropie/configs - if you are certain you don't need a platform, just remove the files yourself. I prefer this over getting complaints that data has been lost. The full uninstall in the main menu asks about 4 times before removing - it used to do it less and someone managed to use it accidentally.

So I'm afraid this is not going to happen for now. If you want to install just 3 emulators, without anything else, just install a fresh retropie-setup instead of using the image.

movisman, 2016-08-01:
No, fair point, I can see why you left it out, and cheers for explaining the reasons for this. I can totally understand the preference to retain data to minimise complaints from users, like you say someone will still manage to erase completely even if it confirms 5 times, I get that. If it is only the configs and roms areas which should be left over, I can just remove these manually. Any other files which might be hanging around elsewhere can just be left.

Out of interest, if installing manually over a raspbian lite image, installing the script and installing only the emulators I need, are there any OS tweaks/adjustments or otherwise, which by default would be missing vs. using the pre-configured image?

joolswills, 2016-08-01:
These are the changes made to a raspbian lite:

https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/admin/image.sh#L83-L107

ran as root:
```
# extra quiet as the raspbian usr/lib/raspi-config/init_resize.sh does
# sed -i 's/ quiet init=.*$//' /boot/cmdline.txt so this will remove the last quiet
# and the init line but leave ours intact
sed -i "s/quiet/quiet loglevel=3 consoleblank=0 quiet/" chroot/boot/cmdline.txt

# hostname to retropie
echo "retropie" >chroot/etc/hostname
sed -i "s/raspberrypi/retropie/" chroot/etc/hosts
```

Then everything else is done via the retropie-setup script from the pi user - you can see in the link above what modules are called (they are called via commandline mode with retropie_packages, but you can do it all from the gui)

You will need to do things like enabling samba shares (if you use them) and so on.

joolswills, 2016-08-01:
Oh, that SED is set to be done BEFORE booting from the image. If booting from it, dont run it, and just instead add "loglevel=3 consoleblank=0 quiet` to /boot/cmdline.txt if you want less terminal output when booting. you can skip this if you dont care.

closed 2016-08-01 by joolswills

movisman, 2016-08-01:
Thanks for that Jools, the above is most useful if I go down the manual install route.

Cheers for the help.

joolswills, 2016-08-01:
np.