RetroPie/RetroPie-Setup #1606: Help with connecting my second iBuffalo SNES controller to retropie 3.8

MOXEZ, 2016-08-02:
Hello, im new to all this Raspberry Pi and RetroPie.

I bought a Raspberry Pi 3 and installed the retropie 3.8. First of all it could only recognize my PS4 controller, but after i ran an update of the firmware/kernel my iBuffalo SNES controller was found and works perfect.

Here comes the real problem, it cant seem to work with my second SNES controller. Not even if i only plug that one in. And i cant run with two at the same time (PS4 and SNES). Can someone please help me fix this problem? My wish is to play 2 player games with my two SNES controllers.

Thank you very much :)

MOXEZ, 2016-08-02:
The weird thing is that if i connect my PS4 joystick and one SNES controller it works just fine. But the problem is to get the second SNES controller to be recognized my retropie. It cant find it even if the second controller is the only one plugged in :/

joolswills, 2016-08-02:
When you opened an issue it shows this text - please use the forum for support.

>This issue tracker is NOT for Help / Support - Use the RetroPie forum - https://retropie.org.uk/forum/

>Please make sure you read the "guidelines for contributing" above before opening an issue. Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes. Please remove this text before posting.

closed 2016-08-02 by joolswills