RetroPie/RetroPie-Setup #1608: Couple of 'cancel' options quit script entirely rather than going back to previous menu

movisman, 2016-08-02:
Hi Jools,

Sorry to keep raising silly little bugs like these, but just spotted the following on a clean 4.0RC1 instance:

- Launch setup script from retropie menu in ES
- Go to manage packages
- Go to manage core packages
- Go to 129 retroarch
- Go to configuration / options
- Click _either_ Manage shaders OR manage overlays (if installed)
- When presented with the two options to install/update or uninstall, click 'cancel'
- Setup script terminates and goes back to ES - expected behaviour is to go back to the previous screen

As a side note, have 'overlays' for retroarch been deliberately left out of the image? I had to install them, maybe you left them off deliberately though - they are about 35mb once installed. Just checking regarding this, just in case as I know they were included previously.

Thanks

joolswills, 2016-08-02:
Thanks. Bit of left over debug code I think. Fixed.

The overlays are not installed by default - as you mention they are 35mb or so, and not everyone uses them, so they are not optional. We used to ship retroarch with all shaders and overlays by default, but you could not then update the shaders or overlays independently and had to wait for a retroarch update or rebuild retroarch from source - now they are managed separately so if a shader is updated, you can update.

closed 2016-08-02 by joolswills

movisman, 2016-08-02:
Great, I thought it might be an easy fix. Updated, tested, no longer a problem.

Thanks again.

commit 1de02412 (2016-08-02): retroarch - remove debug code - fixes #1608