RetroPie/RetroPie-Setup #1609: Add snes9x/pisnes config script

superjamie, 2016-08-03:
Currently native snes9x and PiSNES do not have their controls configured in `/opt/retropie/configs/snes/snes9x.cfg` by the ES configscripts.

I intend to fix this, just logging an Issue to track it.

joolswills, 2016-08-03:
Thanks. You can probably use the pifba one as a basis (similar config). The pifba sdl1_map array can be moved to inputconfiguration.sh and used by both probably also.

superjamie, 2016-08-06:
I thought about moving the SDL keyboard map into `inputconfiguration.sh` but not every emulator uses SDL key mappings for the keyboard.

It seems more appropriate to put the SDL keymap into maybe `$rootdir/lib/`, then input config modules can `source` that keymap file during their `onstart_$module_keyboard` as required.

Thoughts?

joolswills, 2016-08-06:
the mapping isn't useful outside of the input configuration, so it's best to include it in a function in the main code or something.

joolswills, 2016-08-06:
just checked the module - looks like the mapping code is identical to pifba - so it could just use the pifba functions as mentioned on the PR. even simpler then.

closed 2016-08-08 by joolswills