RetroPie/RetroPie-Setup #1611: Bluetooth pairing bug

jkiddo, 2016-08-04:
Im not really sure how much this specifically relates to retropie, but the setup script chokes when pairing to a mobile phone over bluetooth:

Traceback (most recent call last):                                             
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/home/pi/RetroPie-Setup/scriptmodules/supplementary/bluetooth/bluez-simple-agent", line 118, in pair_reply
    dev_connect(dev_path)
  File "/home/pi/RetroPie-Setup/scriptmodules/supplementary/bluetooth/bluez-simple-agent", line 38, in dev_connect
    dev.Connect()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.Failed: No such file or directory

joolswills, 2016-08-04:
The config tool is not designed to pair with a mobile and I've never tested it with that. You don't mention what pairing options you used - but you would probably be better trying to use bluetoothctl from commandline if the config tool doesn't work. 

Did you post to the forum first as requested when opening a ticket ?

joolswills, 2016-08-04:
I tried pairing with my phone and I can at least see one issue that can be worked around, as there is a response we don't handle. I will look into it. I assume you have some app to use the phone as a bluetooth gamepad or something ?

joolswills, 2016-08-04:
I can get my phone to pair but it still throws an error like above - it may just be incompatible with the scripts we use to communicate with bluez over dbus. You have have more luck with bluetoothctl as mentioned.

joolswills, 2016-08-04:
I have made a code change that may help or not - if you update retropie-setup.

closed 2016-08-08 by jkiddo