RetroPie/RetroPie-Setup #1618: xboxdrv - cannot activate 4 gamepads

benparkes, 2016-08-06:
Hey,
Thanks for building such a beautifull system, it really is great!
I have having issues getting 4x controllers to work using the setup script.
I launch the setup script, and select setup>832 xboxdrv and then select 3 for number of controllers.
When i select 4 controllers, then select cancel and reboot from the main menu of the script it does not apply.
each time i run the script it always says 2 controllers
i have tried disabling>reboot>set controllers>enable but still does not change anything.
On antother note - 2 controllers are xbox one and 2 controllers are xbox 360 but i think this issue is before the controllers are even reognised :)
thanks!

joolswills, 2016-08-06:
please use the forum first for support as mentioned when opening a ticket.

closed 2016-08-06 by joolswills

joolswills, 2016-08-06:
The script will always show 2 controllers when launched. You set it to 4, then configure and it will do the set-up for 4 - it just doesn't save those settings to know what they are for next launch.

joolswills, 2016-08-06:
There are also known issues with xboxdrv currently, and it may not work with your xbox one controllers also.

benparkes, 2016-08-07:
yes that is the case now -  i had to uninstall/reinstall for the script to apply for 4 controllers but now i think i need to downgrade the kernel for the driver to work. I previously had 2xXbox 1 controllers working before i reapplied so i think i can get it working. thanks for the advice :)