RetroPie/RetroPie-Setup #1630: Auto Save State Archiving - Feature Request

thorst, 2016-08-11:
I was lamenting how easy it is to overwrite your save states with a friend. We both looked at each other and immediately had the same idea. I did some google searches and there are a bevy of solutions involving cron nightly backups. For ease of use/implementation many suggest settings retroarch to centralize the save states directory. However, to me this isn't elegant as you can progress significantly in a days time.

Writing-
It would be nice to detect a state being written and then backup it up immediately. Im not sure if you can tie into the controller keypress or save state event. If not you would have to use a file system watcher (https://www.infoq.com/articles/inotify-linux-file-system-event-monitoring).

Files set-

- You could use a ever increasing number that increments by 1. *.state0.0 *.state0.1 .... *.state0.3243
- You could use epoch *.state0.1470913540
- You could use a 0-9 suffix that would require each save to rename existing files. so at first it writes .0, then on next save it renames .0 to .1 and then it creates a new .0

The first option seems best here

Cleanup-
I would think you would default to 5 or 10. The chances of you by accident saving 5 times is a stretch I think. For the 3rd option above you wouldn't need to check how many files existed. That would be part of the steps in creating the files. For options 1 and 2 you would need to sort the name and then delete the oldest (top) one. In this way, epoch would be easier because it will sort better out of the box, due to being a portable string (think .0, .1, .10, .2).

joolswills, 2016-08-11:
it sounds a bit over engineered to me, and not sure it is something we would want to include by default - feel free to make a third party script for users though. FOr ertroarch changes you would need to feed them back to the retroarch project as we don't maintain that.

thorst, 2016-08-11:
I just asked over at retroarch. They have a "Undo Save State" feature. Is this exposed in retropie, if so how do I get to it? Is there a way to make this a shortcut on my controller?

joolswills, 2016-08-11:
I have no idea - you could ask on the retropie / retroarch forum - this is the PR https://github.com/libretro/RetroArch/pull/3069

dankcushions, 2016-08-11:
you can use the undo save state stuff in retroarch in retropie if you update retroatch via source. it's in the quick menu in the RGUI. this issue section should not be used for support requests, though.

closed 2016-08-14 by thorst