RetroPie/RetroPie-Setup #1632: [BUG] "name_of_emulator.so" not found

LodanZark, 2016-08-12:
If the network gets disconnect in middle of installation of emulator from the source, leaves the compile files on the directory /home/pi/RetroPie-Setup/temp/build/*  which cause the error of "name_of_emulator.so" not found" after retrying the process.

This problem can be fixed by assure the folder /home/pi/RetroPie-Setup/temp/build/ gets clean before the compiling process.

joolswills, 2016-08-12:
If a build fails it should be cleaned so I need an actual example with a log please.

joolswills, 2016-08-12:
Also you need to include things like what version of retropie-setup you are running.

LodanZark, 2016-08-12:
The log I've got before I manually clean the folder  /home/pi/RetroPie-Setup/temp/build/ during the compiling lr-picodrive form source. 
I'm using the retropie 4.0rc1 script (latest)

Log started at: Fri 12 Aug 15:52:07 UTC 2016
/home/pi/RetroPie-Setup/tmp/build/lr-picodrive /home/pi

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-picodrive' : Sega 8/16 bit emu - picodrive arm optimised libretro core
= = = = = = = = = = = = = = = = = = = = =

fatal: index file smaller than expected
fatal: No such ref: HEAD
fatal: index file smaller than expected
fatal: index file smaller than expected
fatal: index file smaller than expected
/home/pi
/home/pi/RetroPie-Setup/tmp/build/lr-picodrive /home/pi

= = = = = = = = = = = = = = = = = = = = =
Building 'lr-picodrive' : Sega 8/16 bit emu - picodrive arm optimised libretro core
= = = = = = = = = = = = = = = = = = = = =

make: *** No rule to make target 'clean'.  Stop.
make: *** No targets.  Stop.
/home/pi
Could not successfully build Sega 8/16 bit emu - picodrive arm optimised libretro core (/home/pi/RetroPie-Setup/tmp/build/lr-picodrive/picodrive_libretro.so not found).

Log ended at: Fri 12 Aug 15:52:08 UTC 2016
Total running time: 0 hours, 0 mins, 1 secs

joolswills, 2016-08-12:
output of `cd ~/RetroPie-Setup; git log -n 1` please

LodanZark, 2016-08-12:
commit dc33ea6c83b8325ade30a0fcc0e16bbc6db4fcb7
Author: Jools Wills <buzz@exotica.org.uk>
Date:   Thu Aug 11 14:07:25 2016 +0100

joolswills, 2016-08-12:
I will make a change to workaround this - we used to clean the source directory before every build, but there are other issues with that when developing modules etc. I will come up with something. we do clean after sources function fails, but we don't catch the git errors, so that should be enough to sort it.

LodanZark, 2016-08-12:
Thank you very much ;)

joolswills, 2016-08-12:
I couldn't actually reproduce any problem though. I interrupted git, and it was fine when retrying

joolswills, 2016-08-12:
Maybe you have to interrupt git at a specific point.

commit dbaccfcd (2016-08-12): make sure git errors are always trapped when updating submodules - so a failure will trigger a clean. fixes #1632
closed 2016-08-12 by joolswills