RetroPie/RetroPie-Setup #1633: [Feature Request] Dolphin for x86 systems

j-r0dd, 2016-08-12:
I would love to see an option for dolphin in maybe the experimental package section for x86 systems. I know the pi definitely could not handle this.

joolswills, 2016-08-23:
This is included now.

closed 2016-08-23 by joolswills

j-r0dd, 2016-08-23:
Awesome! BTW the hotkeys get stored in ~/.config/dolphin-emu/Hotkeys.ini. Here's what mine look like after only changing select + start to stop the emulator:

`[Hotkeys1]                                                                                                                                                                                      
Device = evdev/0/PLAYSTATION(R)3 Controller
Keys/Open = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & (`Control_L` | `Control_R` )) & O
Keys/Toggle Pause = `F10`
Keys/Stop = `Button 0` & `Button 3`
Keys/Toggle Fullscreen = (`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & Return
Keys/Take Screenshot = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F9`
Keys/Connect Wiimote 1 = (`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F5`
Keys/Connect Wiimote 2 = (`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F6`
Keys/Connect Wiimote 3 = (`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F7`
Keys/Connect Wiimote 4 = (`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F8`
Keys/Connect Balance Board = (`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F9`
Keys/Disable Emulation Speed Limit = Tab
Keys/Freelook Decrease Speed = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `1`
Keys/Freelook Increase Speed = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `2`
Keys/Freelook Reset Speed = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & F
Keys/Freelook Move Up = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & E
Keys/Freelook Move Down = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & Q
Keys/Freelook Move Left = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & A
Keys/Freelook Move Right = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & D
Keys/Freelook Zoom In = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & W
Keys/Freelook Zoom Out = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & S
Keys/Freelook Reset = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & R
Keys/Load State Slot 1 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F1`
Keys/Load State Slot 2 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F2`
Keys/Load State Slot 3 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F3`
Keys/Load State Slot 4 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F4`
Keys/Load State Slot 5 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F5`
Keys/Load State Slot 6 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F6`
Keys/Load State Slot 7 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F7`
Keys/Load State Slot 8 = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F8`
Keys/Save State Slot 1 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F1`
Keys/Save State Slot 2 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F2`
Keys/Save State Slot 3 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F3`
Keys/Save State Slot 4 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F4`
Keys/Save State Slot 5 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F5`
Keys/Save State Slot 6 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F6`
Keys/Save State Slot 7 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F7`
Keys/Save State Slot 8 = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F8`
Keys/Undo Load State = (!`Alt_L` & !(`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F12`
Keys/Undo Save State = (!`Alt_L` & (`Shift_L` | `Shift_R`) & !(`Control_L` | `Control_R` )) & `F12``

Also in ~/.config/dolphin-emu/Dolphin.ini `KeepWindowOnTop = True` under [Display] will fix the issue where you need to click the mouse on the window with the rom running. Really though scripting everything to get this completely up and running may be difficult especially since the graphics settings need to be fine-tuned per system.

gizmo98, 2016-08-23:
@j-r0dd 
Thank you very much. This info is really cool. I will take a look if some kind of controller config can be implemented.