RetroPie/RetroPie-Setup #1636: grep error message after playing libretro ports

ProxyCell, 2016-08-14:
I am seeing the following error message printed on my screen (very briefly) after playing lr-tyrquake/nxengine/dinothawr ports mentioned in the subject. Both ports are LR:
`grep: /opt/retropie/configs//retroarch.cfg: No such file or directory`
`grep: /opt/retropie/configs/quake/retroarch.cfg: No such file or directory`
`grep: /opt/retropie/configs/cavestory/retroarch.cfg: No such file or directory`
`grep: /opt/retropie/configs/dinothawr/retroarch.cfg: No such file or directory`

Please note the TWO slashes between configs and retroarch in the first one I posted as I caught that one on video.

Otherwise everything is fine but this could be a sign of something being misconfigured.

joolswills, 2016-08-14:
I think I know what the issue is, but it helps if you mention what version of retropie-setup you are running when reporting issues.

closed 2016-08-14 by joolswills

ProxyCell, 2016-08-15:
you're right, my apologies. i'm always using the latest everything, script, modules, etc.

commit 266d73e9 (2016-08-14): runcommand - fix path for retroarchIncludeToEnd. fixes #1636