RetroPie/RetroPie-Setup #1637: Openbor not work

saitoh00, 2016-08-16:
Openbor dont work, only go to menu later select game, but when select start, the games freeze. tested with various games (Golden axe remake, double dragon) with same results

joolswills, 2016-08-16:
probably this - https://retropie.org.uk/forum/topic/535/openbor-issues/2

Hence it is in experimental :) I've never tested it myself though. Probably not much can be done currently.

joolswills, 2016-08-16:
Right now opens are leave it as is (I assume it works with some games, but you might need to find the right versions), or remove it.

saitoh00, 2016-08-21:
Dont work with all games i try

superjamie, 2016-09-10:
This apparently worked before in RetroPie 3.x but it doesn't work now in 4.x.

* https://retropie.org.uk/forum/topic/535/openbor-issues
* https://www.reddit.com/r/RetroPie/comments/52151x/cant_get_openbor_working_missing_files/

RetroPie uses an old version of the OpenBOR source (r3016) because nobody who's tried (including @dankcushions and @zerojay) can get the current official source (r3748) compiling within RetroPie.

I don't know why the previously-working version doesn't work anymore. I tried recompiling from source but that didn't work.

I tried using PAKs from PSP downloads with file timestamps around 2005-2007 but that didn't work either.

Someone on the official forums suggests there's a bunch of SDL joystick work which needs to occur in the source for it to happen and the official devteam is too busy with other things.

* http://www.chronocrash.com/forum/index.php?topic=2190.0

It appears the official project isn't even releasing source anymore. Their SourceForge is a year out of date and there are only x86/Wii/Android binaries (r4153/r4161) on the official forums. The official forums reference the outdated SourceForge and source is not distributed with binaries.

I suggest we remove this port because it doesn't work anymore and a good experience is obviously not going to happen without updated source.

joolswills, 2016-09-10:
I have the latest code compiled / running - I just need to look into some issues with getting it to work fullscreen (although even in current state it may be better than the existing build).

saitoh00, 2016-09-10:
Yeah, talk with the team for upload your code like experimental and with this in future can update it

Darknior, 2017-02-03:
For my part i try many PAK release before 2014 and they works very fine 😃 
I will try to release a package over the net with all the best of them 😆
My only problem was for some games, when they are 16/9 like : 
- Final Fight Gold
- Super Mario Brawl
etc ...
They don't fit all the screen.
And the option "Fullscreen Type" : "Stretch to Screen" doesn't work fine 😢 
Is it possible to fix only this problem please ?
Thanks a lot