RetroPie/RetroPie-Setup #1639: Add mk_arcade_joystick_rpi as an additional driver

superjamie, 2016-08-16:
mk_arcade_joystick_rpi is a kernelspace 2-player GPIO input driver:

https://github.com/recalbox/mk_arcade_joystick_rpi

RecalBox ships with this. It's well documented. It's a good way to get 2-player GPIO up and running on the Pi without extra hardware. It would help people who wish to migrate or switch between RecalBox and RetroPie.

If you're happy to accept the enhancement I'll have a go at writing the setup script module, it should be fairly similar to the existing gamecon modules.

joolswills, 2016-08-16:
Please do. The existing gamecon module is not a good example probably though - or at least it needs work (It should have a menu for adding/removing etc, and the gui function itself should not be doing the work. It's an old module and I have only done some cleanups here and there.

superjamie, 2016-08-16:
Cool. Noting down this that you mentioned on IRC: https://github.com/recalbox/mk_arcade_joystick_rpi/issues/23#issuecomment-186340514

commit 6b2dfcc7 (2016-10-25): added mkarcadejoystick module for building/installing https://github.com/recalbox/mk_arcade_joystick_rpi gpio driver - fixes #1639
closed 2016-10-25 by joolswills