RetroPie/RetroPie-Setup #1642: Image packaging method loses capabilities and other xattrs

superjamie, 2016-08-19:
The `ping` command from `iputils-ping` uses `cap_net_raw+ep` to allow unprivileged users to run the command. It seems the RetroPie image packaging method erases caps and other xattrs.

For `iputils-ping` it's an easy package reinstall, but I am concerned we are missing other xattrs or caps which might break something important one day.

Reference post: https://www.reddit.com/r/raspberry_pi/comments/4yg77g/downloading_libraries_using_noobs_and_pip3/d6njz9o

joolswills, 2016-08-19:
It's not a bug in the RetroPie image packaging - it's a bug in Raspbian (or their image packaging), which our images are based off.

https://www.raspberrypi.org/forums/viewtopic.php?p=853440#p853440

I will put a workaround in for our image building - but it really needs to be fixed upstream - I'm guessing they will look into it for the next release.

joolswills, 2016-08-19:
Actually - you might be right, i had assumed it was because raspbian used to have this bug, but I think their latest image solves it. Will look into it.

joolswills, 2016-08-19:
Thanks for this - I had assumed `rsync -a` preserved xattrs also, but it doesn't - will add the additional parameters.

commit 6c2b6ce5 (2016-08-19): image - preserve everything when syncing filesytem hardlinks/xattrs/acls - thanks superjamie - fixes #1642
closed 2016-08-19 by joolswills