RetroPie/RetroPie-Setup #1643: 8bitdo contoller stops working when PS3 driver installed

swd120, 2016-08-19:
This is with an 8bitdo NES30 controller paired in start+r mode - I don't know if it affects other 8bitdo controllers, but it very well could.

Replication steps:  

1. Pair 8bitdo NES30
2. Test controller in game/emulation station (should work fine)
3. Install PS3 input driver via Retropie-Setup
4. Try to using 8bitdo controller in EmulationStation/or a game - It will fail to register inputs
5. Uninstall PS3 input driver via Retropie-Setup
6, Test controller in game/emulation station (should work fine)

This is frustrating, and I'd like to use NES30 controllers for older games, and PS3 controllers for N64/PSX/Dreamcast etc.

gizmo98, 2016-08-19:
Sorry. You cannot use ps3 controllers and other bluetooth controllers at the same time. The ps3 sixad driver disables and replaces the bluetooth daemon.
https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/ps3controller.sh#L83

closed 2016-08-19 by gizmo98

joolswills, 2016-08-19:
Also - please always post to the forum first as detailed when opening a ticket. When pairing a ps3 controller there is a big warning now on the current retropie that details this. Some people have managed to pair their devices first, then install the ps3 driver afterwards and have said their devices still pair, however this is "untested".

joolswills, 2016-08-19:
(and it doesnt seem to work in your case)

swd120, 2016-08-19:
The device still pairs - the inputs don't register properly though.

I don't quite understand why you would close this as an issue though...  Seems like an outstanding problem that should be noted in the issue log, even if its not something that can be fixed in a short timeframe...

joolswills, 2016-08-19:
We are not responsible for every piece of software we ship with - however as I wrote above when you go to pair a PS3 controller, a window opens up stating this incompatibility.