RetroPie/RetroPie-Setup #1646: [4.0.1] RPi2 - Xpad can't install

nemo93, 2016-08-21:
I've decided to upgrade kernel to 4.4.0 as I'm now relying on the metapackage `linux-image-rpi2-rpfv` for that. Then I've read a bit about Xpad and as suggested in the Wiki I decided to give it a try. I launched RP menu for that. However install fails right at the beginning as the script tries to fetch an apparently non-existing `raspberrypi-kernel-headers` package.

Even after installing `linux-headers-rpi2-rpfv` the problem remains. Also if I try to install `dkms` package there's a long list of old 3.6 kernels and gcc/cpp that are to  be installed which I'd like to avoid if possible.

Here's trace from RP menu logs:
```
~/RetroPie-Setup $ zless logs/rps_2016-08-21_141248.log.gz
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Hit http://archive.raspberrypi.org wheezy/main armhf Packages
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en_GB
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en
Ign http://archive.raspberrypi.org wheezy/main Translation-en_GB
Ign http://archive.raspberrypi.org wheezy/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package raspberrypi-kernel-headers
Could not install package(s): dkms raspberrypi-kernel-headers.

Log ended at: Sun 21 Aug 14:13:40 CEST 2016
Total running time: 0 hours, 0 mins, 47 secs
``` 
I'm using RP latest update and I'm on Jessie (8.0) running up-to-date packages. Any other info please let me know.

joolswills, 2016-08-21:
I would prefer things like this posted to the forum first as mentioned when opening a ticket -  however this is not a bug as such as it is designed to build with raspberrypi-kernel-headers - which is available in the archive.raspberrypi.org archive - so you are possibly missing this repository. 

What is linux-image-rpi2-rpfv required for - Raspbian doesn't need that for kernel 4.4. - are you not running Raspbian ?

joolswills, 2016-08-21:
dkms is required unless you want to build it manually - which you will have to work out for yourself. I dont remember it having dependencies on any old kernels though.

nemo93, 2016-08-21:
Argh apologize once again for not using forum for that. I swear I'll do that next time. 

Wrt to the issue in fact I do use a very old RP version (perhaps from 2.0??) and upgraded it regularly (packages). Therefore I'm on Raspbian but I haven't the "archive" repository in my `sources.list` indeed... Adding it right now and continue testing. Sorry about that.

nemo93, 2016-08-21:
Should I close that ticket and continue on the forum btw?

Still failing:

````
= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'xpad' : Updated Xpad Linux Kernel driver
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): dkms raspberrypi-kernel-headers. I am trying to install them now.
Hit http://archive.raspbian.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://raspberrypi.collabora.com wheezy InRelease
Hit http://archive.raspberrypi.org wheezy InRelease
Hit http://archive.raspbian.org jessie/main armhf Packages
Hit http://archive.raspbian.org jessie/contrib armhf Packages
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Hit http://archive.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Hit http://archive.raspbian.org jessie/rpi armhf Packages
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Hit http://raspberrypi.collabora.com wheezy/rpi armhf Packages
Hit http://archive.raspberrypi.org wheezy/main armhf Packages
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en_GB
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en
Ign http://archive.raspberrypi.org wheezy/main Translation-en_GB
Ign http://archive.raspberrypi.org wheezy/main Translation-en
Ign http://archive.raspbian.org jessie/contrib Translation-en_GB
Ign http://archive.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://archive.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://archive.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://archive.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://archive.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://archive.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://archive.raspbian.org jessie/rpi Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package raspberrypi-kernel-headers
Could not install package(s): dkms raspberrypi-kernel-headers.
```

Content of my sources.list
```
~/RetroPie-Setup $ cat /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
deb http://archive.raspbian.org/raspbian jessie main contrib non-free rpi
```
Current Raspbian
```
~/RetroPie-Setup $ cat /etc/debian_version
8.0

```

nemo93, 2016-08-21:
In case that helps output when I try to install `dkms` by hand.
```
~/RetroPie-Setup $ sudo apt-get install dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  cpp-4.6 gcc-4.6 gcc-4.6-base libuuid-perl linux-base
  linux-headers-3.6-trunk-common linux-headers-3.6-trunk-rpi
  linux-image-3.6-trunk-rpi linux-kbuild-3.6
Suggested packages:
  gcc-4.6-locales libmudflap0-4.6-dev gcc-4.6-doc libgcc1-dbg libgomp1-dbg
  libquadmath-dbg libmudflap0-dbg libcloog-ppl1 libcloog-ppl0 libppl-c4
  libppl12 libppl9 linux-doc-3.6 debian-kernel-handbook fdutils
Recommended packages:
  linux-headers-686-pae linux-headers-amd64 linux-headers-generic
  linux-headers linux-image
The following NEW packages will be installed:
  cpp-4.6 dkms gcc-4.6 gcc-4.6-base libuuid-perl linux-base
  linux-headers-3.6-trunk-common linux-headers-3.6-trunk-rpi
  linux-image-3.6-trunk-rpi linux-kbuild-3.6
0 upgraded, 10 newly installed, 0 to remove and 0 not upgraded.
Need to get 14.8 MB/23.5 MB of archives.
After this operation, 87.8 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

joolswills, 2016-08-21:
You don't have the right repositories installed - You have upgraded from wheezy I guess and missed some steps (which is why they recommend you start from a fresh jessie instead of upgrading on the raspberrypi.org forums)

As I mentioned above you don't have the archive.raspberrypi.org repository.

You also have old wheezy repositories you should remove.

dkms does not install cpp-4.6 on jessie - it's not even available - it's a wheezy package

I recommend you start from a new Raspbian.

closed 2016-08-21 by joolswills