RetroPie/RetroPie-Setup #1648: Autostart ROM on bootup

MichelFreund, 2016-08-22:
My arcade cabinet, which I use Retropie for, should resemble an old arcade game as much as possible. Therefore, I would like to have Retropie be able to autostart a specific ROM on bootup.

This should be possible by manipulating 10-emulationstation.sh, but I would rather like to be able to configure it from a menu so that I can change it easily.

Autostart would imply a ROM is started after bootup, but if you would escape the ROM you would end up in Emulationstation.

joolswills, 2016-08-23:
>This issue tracker is NOT for Help / Support - Use the RetroPie forum - https://retropie.org.uk/forum/

>Please make sure you read the "guidelines for contributing" above before opening an issue. Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes. Please remove this text before posting.

closed 2016-08-23 by joolswills

MichelFreund, 2016-08-23:
Hi Jools,

In fact, this request was not meant as a help request. What I ask for, doesn’t exist yet in the GUI.
I see a lot of questions on forums in which users ask how they should setup an autostart Till now, this can only be arranged using scripts. I would like to see a menu item.

Kind regards,

Michel

joolswills, 2016-08-23:
It would still be best to post to the forum first to get feedback from the devs or else this issue tracker would be full of feature requests. I don't think this is something common enough to warrant a gui - and it's simple enough to add it manually.