RetroPie/RetroPie-Setup #1652: Can't install on RaspberryPi 3 - B (Ubuntu Mate)

k0st1x, 2016-08-24:
i have rpi3b with ubuntu mate
install on it libsdl1.2-dev & libsdl2-dev packages
clones retropie-setup repository and tried to execute "basic install"
after some hours dialog shows me messages "Unable to install binary for sdl2" and "Unable to install binary for sdl1"
attaching logs...
[rps_2016-08-21_222846.log.zip](https://github.com/RetroPie/RetroPie-Setup/files/435399/rps_2016-08-21_222846.log.zip)
in logs can be found that libsdl was trying to install from sources, but I think it was unnecessary.
Can some one help me to describe issue that retropie can not be installed on ubuntu mate 16.04?

joolswills, 2016-08-24:
Ubuntu on the RPI is not supported - you will need to use Raspbian.

joolswills, 2016-08-24:
I have changed the code so it will detect this combination and throw an error - on the Raspberry Pi we rely on specific packages for Raspbian - work would be needed to make it work on Ubuntu on the RPI, and I don't have the time to do that (or support it).

k0st1x, 2016-08-24:
ok. i understand that.

closed 2016-08-24 by k0st1x