RetroPie/RetroPie-Setup #1659: Limelight discovery quit causes script to quit

smartroad, 2016-08-27:
I have been trying to play with Limelight but have an issue when it goes to the discovering 'GeForce PCs'. It says to hit CTRL-C to stop the search, but when you do it actually quits the script completely. Otherwise it just sits there for ages apparently not doing anything :(

I can't uninstall it either for the same reason - I wanted to do an update of all parts but this just stops it dead

joolswills, 2016-08-27:
You are probably better off using moonlight - which isn't yet implemented automatically. see https://github.com/RetroPie/RetroPie-Setup/issues/1082

However I will see if the ctrl+c can be fixed up.

smartroad, 2016-08-27:
Cool thanks, I'll look at moonlight. :)

joolswills, 2019-09-11:
Closing as limelight is now replaced with moonlight.

closed 2019-09-11 by joolswills