RetroPie/RetroPie-Setup #1660: Add support for PS3 USB Versus fighting controller

kentonbmax, 2016-08-27:
Please add support for this controller in the next version of retroarch. I currently will not work with any emulators on PIE 4.0. Considering switching to recalbox to see if it works. Controller support is a major downside to the usability of of retroarch/retropie; as users spend more time trying to make them work then actually playing games. Going to try the work around for retrolink's n64 controller. Thing is, works great on pc. Cheers!

joolswills, 2016-08-27:
have you opened a forum thread in regards to getting this working (as requested when opening an issue) ? Mostly we rely on upstream kernel support - RetroPie is not a standalone OS, but runs on top of Linux.

kentonbmax, 2016-08-27:
Trying to understand this properly, the controller works fine in emulation
station, its when playing games that the usability process breaks down. I
interpreted the setup to mean that it includes setting up inputs. The forum
is the only place to address controller issues? Would this be retroarch or
retropie issue?

joolswills, 2016-08-27:
It may be an issue with the autoconfiguration, or just a config issue but it should be debugged first on the forum please, then a ticket can be opened if needed.

kentonbmax, 2016-08-28:
Done. Where is the best place to communicate enhancements?

joolswills, 2016-08-28:
Everything on the forum first please.

closed 2016-08-28 by joolswills