RetroPie/RetroPie-Setup #1666: neo geo pocket in game save problem

dego93, 2016-08-29:
The emulator creates the flash save when exit the game but when i reload the game the save won't load.
Tested games Biomotor unitron, evolution and dark arms.
I have already remove and reinstall the emulator lr-beetle-ngp.

dego93, 2016-09-01:
I did a fresh install with the 4.0.2 img on another sd but the neo geo pocket and neo geo pocket color save don't work.
I downloaded retroarch 1.3.6 for windows and the core mednafen_ngp_libretro.dll (i don't find lr-beetle-ngp) and the in game save work good.

joolswills, 2016-09-01:
Read the text again that you removed when opening an issue.

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

closed 2016-09-01 by joolswills