RetroPie/RetroPie-Setup #1676: Retropie is missing libSDL_mixer-1.2.so.0

LeLobster, 2016-09-06:
libSDL_mixer-1.2.so.0 is required for both EDuke32 and Giana's Return, and possibly others, both installable via the Retropie Setup.

Running fully updated Retropie 4.0
Installed both ports from binary via retropie setup but got this error message when trying to start

`error while loading shared libraries: libSDL_mixer-1.2.so.0: cannot open shared object file: No such file or directory`

Looked up the package for libSDL_mixer for Jessie on debian.org
ran `sudo apt-get install  	libsdl-mixer1.2` and that fixed it.

joolswills, 2016-09-07:
eduke32 already includes the dependency - so that should work. I will add it to Giana's return along with other dependencies

joolswills, 2016-09-07:
eduke32 actually contains dependency for libsdl2-mixer-dev - and it should be building against sdl2 - unless it is mixing both sdl2 and sdl1 libraries but I will check.

joolswills, 2016-09-07:
Can confirm the prebuilt binary for eduke32 is built against libsdl1.2 mixer.

biscuits99, 2016-09-07:
Looks like its needed here as well, although this isnt part of the standard install so more an FYI than anything else

https://retropie.org.uk/forum/topic/3432/is-streets-of-rage-remake-working-for-anyone-in-4-0/11

joolswills, 2016-09-07:
That will need reporting to zerojays repo

biscuits99, 2016-09-07:
Thanks, now done.

joolswills, 2016-09-07:
I have rebuilt the eduke32 binaries and it is now correctly built against libsdl2-mixer

commit 7dba1d45 (2016-09-07): giana - added missing dependencies - fixes #1676
closed 2016-09-07 by joolswills