RetroPie/RetroPie-Setup #1677: Shanwan PS3 Controllers not working with Retropie 4

Prindle19, 2016-09-07:
As discussed in the [forum](https://retropie.org.uk/forum/topic/3472/shanwan-ps3-support-since-installation-of-retropie-4-0)

retropiehd[4059] Authentication attempt without agent
retropiehd[4509] access denied org.bluez.Error.Rejected
retropiehd[4509] 01:B5:8E:82:63:25: error updating services: Connection timed out (110)

In the syslog when I try to pair a Shanwan controller that has been previously connected via usb.

joolswills, 2016-09-07:
@gizmo98 do you have one of these controllers ?

joolswills, 2016-09-07:
aah hmm, it might be to do with the way we call the module in retropie 4 - will look into it.

joolswills, 2016-09-07:
It doesn't allow the others to be installed currently - working on a fix.

gizmo98, 2016-09-07:
I have one but did not test with rp4.

joolswills, 2016-09-07:
@gizmo98 while fixing this, I was wondering about the meaning of "clone support gasia" "clone support shanwan" does this mean "driver with controller clone support including shanwan" or something else ?

gizmo98, 2016-09-07:
If i remember correctly clone support gasia allows one original or one gasia controller. Gasia only allows multiple gasia controllers but no original ones.

joolswills, 2016-09-07:
Please update retropie-setup - remove the ps3 controller and re-install. On the initial install it will install the driver for the official ps3 controllers - but then you can use the configuration / gui to switch to another.

Prindle19, 2016-09-08:
@joolswills yep, that worked just fine.

closed 2016-09-08 by joolswills

ramsbod, 2016-12-31:
I have the same controller and got it to work via USB cable but can't get Bluetooth connection working.  Did you get the Bluetooth working? I have Pi 2 with Bluetooth dongle.

gordogato, 2017-03-24:
@ramsbod do you still need help getting your controllers working? i have no idea how i did it but I got it working and can even transfer it to another pie. i have 4.1.19 working that i can copy and paste to the new install and it works. (to clarify, i copy the ENTIRE root directory from the working pi to the non-working pie.... then it works )

allawi951, 2018-07-01:
I have PS3 Shanwan Controller & I tried to update retropie setup and re-install ps3 controller & change it to clone shanwan but my controllers can't pair, only 4 led flashing after plug out the usb for some seconds then switched off.

any ideas how to pair my controlers? I have retropie 4.4

joolswills, 2018-07-01:
This issue is closed - please use the forum.